DacoTaco/priiloader #311: "Loading disc" not working

PownicMania, 2022-08-12:
**Describe the bug**
I tried to load my Mario Kart Wii disc through the "Load disc" feature in Priiloader twice, yet it didn't load the game at all. The first time even had a glitched green screen.

**To Reproduce**
Steps to reproduce the behavior:
Have Mario Kart Wii on your disc drive
Load it via Priiloader

**Version**
0.10.0 Beta 1

**Expected behavior**
Load the game successfully

TheTechRobo, 2022-08-12:
Could you enable logging and post the Prii.log file?

DacoTaco, 2022-08-13:
and on top of that log, does the game work when loaded from system menu/another source?

PownicMania, 2022-08-14:
> and on top of that log, does the game work when loaded from system menu/another source?

It works just fine on the system menu

> Could you enable logging and post the Prii.log file?

How do you enable logging? I don't see it anywhere

DacoTaco, 2022-08-14:
it should be in the settings, i think its name is "dump debug info" or something along the lines.

sadly i don't own a copy of mario kart on the wii to try this out with :/ 
@TheTechRobo  : do you have the game, and can you try?

TheTechRobo, 2022-08-14:
I have Mario Kart Wii for NTSC. I do not currently have Priiloader 0.10 beta. Despite being a beta, is it safe for me to install it? (Also, is there an installer binary I can use or just the source code)?

DacoTaco, 2022-08-14:
hm, beta seems safe to install but if youre that worried, maybe its better to try a priiloader build in HBC?
run the dol in the following zip in HBC, see if it works?
[priiloader.zip](https://github.com/DacoTaco/priiloader/files/9333498/priiloader.zip)

DacoTaco, 2022-08-16:
@PownicMania : did you find the log setting/file?

DacoTaco, 2022-09-04:
i got myself a copy of mario kart wii (PAL) and have confirmed that it indeed crashes. looking at this i *think* its overwriting priiloader when launching the game

[image]

DacoTaco, 2022-09-05:
it was a video buffer/entrypoint issue.
Priiloader has now been moved to 0x81330000, the same address where system menu resides, and the memory region used by loaders.

the problem should be fixed, my copy booted correctly now

closed 2022-09-05 by DacoTaco