DacoTaco/priiloader #312: System Menu 4.X booting broken by bf19a47

AndrewPiroli, 2022-08-18:
**Describe the bug**
Attempting to boot System Menu 4.3U results in a black screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Install IOS80 and System Menu 4.3
2. Install priiloader
3. Try to load the system menu either via autoboot or manually

**Version**
Introduced in commit bf19a47

**Expected behavior**
The System Menu should be loading

**Screenshots**
Imagine a black screen where the System Menu should be.

**Additional context**
Reproducible in Dolphin
I am gathering additional information with a debug build right now. I will update soon with more information.

AndrewPiroli, 2022-08-18:
Built priiloader with this patch (I did not include the hack to gprintf to always dump gecko to SD or \#define DEBUG).
[0001-Increase-debug-factor.patch.txt](https://github.com/DacoTaco/priiloader/files/9367602/0001-Increase-debug-factor.patch.txt)

It logs the calculated content fileID both how it was before and after the commit in question. And the resulting NAND load path.
And the output from the enhanced logging: [prii.log](https://github.com/DacoTaco/priiloader/files/9367614/prii.log)

The log shows the old method of calculating fileID == 0x10000098 vs the method introduced in this commit == 0x000098
Because of the change, Priiloader is trying to load: /title/00000001/00000002/content/00000098.app but on 4.3 Priiloader Installer moves the System Menu to 10000098.app
00000098.app does exist but it is very small (is this the loader?) which is why the 'boot file not found' failure case is not triggered.

DacoTaco, 2022-08-18:
ye, i woke up yesterday with that realization that the '&= 0x1FFFFFFF' was wrong and it should be '= 0x10000000 | 0x0FFFFFFF'

DacoTaco, 2022-08-18:
fixed this real quick. thanks for reminding me andy 😸

closed 2022-08-18 by DacoTaco