DacoTaco/priiloader #316: But the 0.10.0???

Leonardo5681, 2022-12-07:
Hi, but for version 0.10.0??? I've been waiting 12 months for the release, because I don't want to use the beta because I don't want to create instability in my Wii. So when is it released?

DacoTaco, 2022-12-13:
sorry for waiting so long.
my plan was to release 0.10 few months after beta 2, but i completely forgot due to personal life and projects.
with vWii support being worked on i might either skip 0.10 or create a 0.10 beta 3 with vwii support

i am not going to release the vwii support in a final release without beta version because it requires a rework in a few critical places.

so stay tuned for 0.10 beta 3, and after that, 0.10 .

if you'd like 0.10 already, im kinda confident 0.10 beta 2 is stable besides a few innocent bugs

closed 2022-12-13 by DacoTaco

Leonardo5681, 2022-12-14:
But do you have discord? Anyway, you're working on it for 0.10.0 and sorry if I wrote to you a little nervous earlier

DacoTaco, 2022-12-14:
no problem, it was a good question. i did forget about my plans, and its good that you reminded me about them.
i do have discord, but i'd rather not have it out there publicly. 
is there a specific feature youre waiting on?

Leonardo5681, 2022-12-14:
No actually no, I just wanted to vent that I've been waiting for 0.10.0 for months and it didn't arrive, anyway, you had completely forgotten and I was the one who reminded you, and then you're starting to work on it again now that you did I remember it?

Leonardo5681, 2022-12-16:
Hey, I'll send you my discord id if you want so you write me without sharing yours