DacoTaco/priiloader #318: Doesn't let you run homebrew app on the vWii on the Wii U

Yeetov, 2022-12-23:
**Describe the bug**
Says "Error: vwii detected - please do not run this on a wii u!"
"Press A to exit back to the loader"

**To Reproduce**
Steps to reproduce the behavior:
Open Priiloader on the vWii Homebrew Channel

**Version**
[0.10.0-beta2](https://github.com/DacoTaco/priiloader/releases/tag/0.10.0-beta2)
[Priiloader 0.9.1](https://github.com/DacoTaco/priiloader/releases/tag/0.9.1)
Haven't tested any others

**Expected behavior**
Install Priiloader onto the vWii on the Wii U

Masamune3210, 2022-12-23:
Release doesnt support vWii yet, afaik. Wait until its mentioned in the release notes before trying it

Yeetov, 2022-12-23:
It says in the description that it is available for vWii: [https://www.youtube.com/watch?v=KLy_4PXanic](https://www.youtube.com/watch?v=KLy_4PXanic)

Yeetov, 2022-12-23:
Oh yeah, it isn't available: [https://github.com/GaryOderNichts/priiloader/releases](https://github.com/GaryOderNichts/priiloader/releases)

Yeetov, 2022-12-23:
Not available yet for vWii

closed 2022-12-23 by Yeetov

Masamune3210, 2022-12-23:
Correction: It says its being WORKED on
[image]