DacoTaco/priiloader #319: About hack No-Delete hack HAXX, JODI, DVDX, DISC, DISK, RZDx

GDX-X, 2023-01-17:
Hello,

I would like to know what exactly does the No-Delete HAXX, JODI, DVDX, DISC, DISK, RZDx hack do?

Because I explain my problem to you

When I quit Wii Fit Plus, I get an error message System files are corrupted

But when I activate this hack with priiloader I no longer have the error

It seems like people are having a similar problem but no one has been able to identify the exact problem for 10 years and maybe this hack will lead me to a lead.

I tried several things like reinstalling all original ios but the problem is still there

I think it's a corrupt save
but I am not sure

I have my wii in boot2 so I can test a lot of things without too much risk :D

Masamune3210, 2023-01-17:
System Menu versions past certain times will automatically delete any channels that have those ID's, as a way to (poorly) combat homebrew. The hack keeps the system menu from doing so

GDX-X, 2023-01-17:
Ok thanks for the info

Do you know what this error can mean because each time I have system files are corrupted I have this line which is added to the logs; 

00000001-00000002 NWC24 error. [-1] iplNwc24Manager.cpp line: 1347

I guess NWC24 Means Nintendo WiiConnect24
I tried deleting the entire "shared2/wc24" folder But without success

DacoTaco, 2023-01-17:
what SM version are you using?

and that error is interesting, but without the source not very useful i think unless you could find it in the SM binary.
i assume the hack has some errors and its making SM crash.

GDX-X, 2023-01-17:
I am using version 4.3E

But in my memory I also had this problem with version 4.1E

DacoTaco, 2023-01-17:
so the problem is not related to priiloader and the hacks?

GDX-X, 2023-01-17:
No, but the hack allows me to no longer get the error when quitting wii fit plus so I wanted to know more about this hack in case it leads me to a clue of this problem.

DacoTaco, 2023-01-17:
i think it disables certain checks it does on titles, which might cause the error. 
are there any homebrew channels besides the homebrew channel installed?

GDX-X, 2023-01-17:
I currently have installed: HBC, USB Loader GX, WiiMc, Snes9GX, VGBAGX FCEUGX, NK2O

DacoTaco, 2023-01-17:
i'd suggest to uninstall one of those channels, besides HBC, and try again.
repeat until the problem is solved or no more homebrew channels are left

GDX-X, 2023-01-17:
I tried but the problem persists I will try to reproduce the bug with dolphin with the synchronized balance board. to have the logs of what can cause this bug

because i tried everything

there are only two possibilities:

1 - Savedata corrupt

2 A hardware problem with some balance board models

Masamune3210, 2023-01-18:
Are you sure your nand isnt just slightly corrupted in some way? If you used a nand dump from a damaged nand, the issue would follow even into Dolphin

GDX-X, 2023-01-18:
I analyzed my nand with Nand inchecker I did not notice anything strange the only thing that does this to me is with wii fit plus when the balance board is synchronized I have not tried other games compatible with the balance board but according to other people it does it too

you can see it on this thread https://gbatemp.net/threads/the-system-files-are-corrupted-error-exiting-wii-fit-plus.197588/page-4

GDX-X, 2023-01-19:
Update: 

When WiiConnect24 is disabled, the problem disappears,

The problem is that I deleted the WC24 folder thinking of a brick of mail but the problem persists

So I summarize the situation all the games that support the balance board when you return to the wii menu if the wc24 is activated I will have a corrupted system file message

So maybe a corrupted driver in the shared1 folder?

DacoTaco, 2023-01-19:
personally im more interested in how you got the corruption. obviously this can't be a nintendo bug, i have a hard time believing that

GDX-X, 2023-01-19:
I too am interested in what could have caused this but I really don't see especially that it does it on games that require the balance board I wonder why only these games

Masamune3210, 2023-01-19:
Have ypu done any modifications to the WC24 system!? Stuff like Riiconnect? Also, do you have any custom or unrecognized IOS? Maybe a sysinfo output would be useful

GDX-X, 2023-01-19:
No I never made any modification of WC24 I never installed riiconenct24 I had this problem also in 2011 According to the logs of the wii

My syscheck: https://pastebin.com/mJUZqgR3

DacoTaco, 2023-01-19:
the only thing i see there are modded ios'

GDX-X, 2023-01-19:
I don't think it comes from ios because I have another wii with the same configuration without having this bug so everyone would have this bug with the basic cIOS of the hack

according to wiibrew in the shared1 folder there are drivers for the wii is i know that if you install
wad, they can drop "drive" files there. maybe one day while installing something a long time ago i corrupted something in there, (i guess) because i hacked my wii from the start of the hacks

I will try to replace these files with new ones

I thought it came from a corrupted save but I copied all my save data from my wii to another to reproduce this bug without success

DacoTaco, 2023-01-19:
the patched IOS80 is not a cios hehe
thats piracy (i figure)

GDX-X, 2023-01-19:
Yes I did 2 one in 2012 but the problem was already present and another recently before testing things

> the patched IOS80 is not a cios hehe thats piracy (i figure)

Yeah :D

GDX-X, 2023-01-20:
ok i have another theory

Maybe wc24 is trying to read a file from a corrupt or unreadable savedata and so I get the corrupted system file

the problem is that I have 105 saves...

But logically if I copy all my saveata from my wii to another I should have this message too in this case?

I will try to do some tests

closed 2023-01-26 by DacoTaco