DacoTaco/priiloader #320: Wii Mini USA Menu Hacks (v4609)

alrpeary, 2023-01-21:
Noticed some hacks in [hacks_hash.ini ](https://github.com/DacoTaco/priiloader/blob/master/priiloader/hacks_hash.ini) have been made for system menu version 4610, the PAL Wii Mini, but not for menu version 4609, the USA Wii Mini.

I'm personally interested in one of these, `[Replace Health Screen with Backmenu]`, but since I have the hardware/menu version I'd be willing to offer my help in implementing/testing more hacks for the USA Wii Mini.

DacoTaco, 2023-01-22:
the hacks are rather easy to port, i just don't have access to a US Wii mini.
its all a matter of taking the hash values, checking if they are in the targeted SM, test if they are correct and adjust where needed.

seeing that hack though i suspect it'll be a case of looking at the original code in the PAL version, and looking for the similar code in the US version.

i can take a dump and look at the code around the place where it patches for you to compare to the US version if that helps?

alrpeary, 2023-01-22:
That would be greatly appreciated. I'm very new to the hacking scene, in fact its only been a week since I've bought my Wii and soft-modded it, but I have a background in computers so I feel I learn quickly.

If you could share that dump too and a little bit of the process, maybe I can figure out how to engineer the other hacks myself too!

alrpeary, 2023-01-29:
I'm also interested in how I can port `[Region Free EVERYTHING]` too! Although that one seems much more complex.

DacoTaco, 2023-02-04:
sorry it took so long for me to reply. 
here are some findings that could help you figure it out for the us version by looking at a hex editor : 
* the EU version of `Replace Health Screen with Backmenu` has 1 instruction with hex `0x8140A118` (`lwz r10, -0x5ee8(0)`) that it patches to `0x8140A180`. first look if this instruction is also in the US version, and if the code (or hex) around it look the same. if it does, chances are all you have to do is change the minimum version of the hack in the hacks.ini. 4.3E on the mini is version 4610, US is 4609.
* don't worry about trying hacks. the hacks are done in memory and don't alter system menu as it is on the wii's nand flash.
* `Region Free EVERYTHING` might be trickier, but should overall be do-able to port if a working copy of the wii-mini exists for Europe. it is split up in 2 parts. the master code, that works for all system menu versions so far and system menu version specific parts. the master is already set up to have SM v4610 as max, so it includes US version, which means you don't have to look at that part.
* just like before, take the hash from the hack for version 4610 and check if it exists in 4609 and is only in there once. if it is, just change the version and test. if its not, you'll have to find the code in the us version and write a patch. but first check if it is in there

DacoTaco, 2023-03-01:
awesome work! i would also test v3.0 or something because there were a lot of changes between 3.x and 4.x since starting 4.x it started to look at new stuff and a lot of memory/instructions changed.

i assume the hacks changes will come after the vwii support has finished?

vaguerant, 2023-03-01:
> i assume the hacks changes will come after the vwii support has finished?

I believe that's the plan, yeah, they will presumably be added to the current WIP PR once vWii is ready and the new hacks have been tested.

Ingunar, 2023-03-01:
@vaguerant is not even bragging about the [Force Boot into Data Management] hack. :)
That hack should enable quick banner brick unbricking for Wii and vWii. 

He did some really awesome work.

We had a lot of discoveries while testing and will share more info after we finish the last batch we are working on currently.

Ingunar, 2023-03-01:
> now if only that video issue could be resolved... :V

Nintendont developers didn't fix it, and some were discovered about 10 years ago. 
Gary is doing magic currently tbh.

We can upload the new `hacks_hash.ini` soon. But also maybe I should create a new note so we can do cross-system testing?

I am editing this one for our current vWii testing only.
Here is the current [note](https://hackmd.io/@iSVplD0kT2WqWUNU8oF1AQ/Hy88-7LRi)

closed 2023-09-08 by DacoTaco