DacoTaco/priiloader #321: How Do I Compile the Source Code of Priiloader?

P0W3, 2023-01-22:
This is not a bug. I am trying to find a way to compile the source code of Priiloader. I am using Visual Studio Code to compile the source code of Priiloader. I have encountered a few errors after trying to find a way to compile the source code of Priiloader.

```
PS C:\priiloader> makefile
makefile : The term 'makefile' is not recognized as the name of a cmdlet, function, 
script file, or operable program. Check the spelling of the name, or if a path was  
included, verify that the path is correct and try again.
At line:1 char:1
    + CategoryInfo          : ObjectNotFound: (makefile:String) [], CommandNotFound  
   Exception
    + FullyQualifiedErrorId : CommandNotFoundException
 

Suggestion [3,General]: The command makefile was not found, but does exist in the current location. Windows PowerShell does not load commands from the current location by default. If you trust this command, instead type: ".\makefile". See "get-help about_Command_Precedence" for more details.
PS C:\priiloader> .\makefile
PS C:\priiloader> C:\priiloader\priiloader
spelling of the name, or if a path was included, verify that the path is correct     
and try again.
At line:1 char:1
+ C:\priiloader\priiloader
+ ~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\priiloader\priiloader:String) [],  
    CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS C:\priiloader> cd C:\priiloader\priiloader
make : The term 'make' is not recognized as the name of a cmdlet, function, script   
file, or operable program. Check the spelling of the name, or if a path was 
included, verify that the path is correct and try again.
At line:1 char:1
+ make
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (make:String) [], CommandNotFoundExce  
   ption
    + FullyQualifiedErrorId : CommandNotFoundException
 
PS C:\priiloader\priiloader> Makefile
Makefile : The term 'Makefile' is not recognized as the name of a cmdlet, function, 
script file, or operable program. Check the spelling of the name, or if a path was   
included, verify that the path is correct and try again.
At line:1 char:1
+ Makefile
+ ~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (Makefile:String) [], CommandNotFound  
   Exception
    + FullyQualifiedErrorId : CommandNotFoundException
 

Suggestion [3,General]: The command Makefile was not found, but does exist in the current location. Windows PowerShell does not load commands from the current location by default. If you trust this command, instead type: ".\Makefile". See "get-help about_Command_Precedence" for more details.
```

If any help is okay, thanks.

Masamune3210, 2023-01-22:
Powershell is telling you exactly what you did wrong and how to fix it.....

DacoTaco, 2023-01-22:
go to the root folder of priiloader and run make. it will pickup the makefile and build if you have devkitpro's devkitPPC installed.

however, its been a while since ive used devkitPPC on a windows machine and i think you'll need to run make from msys? just a note incase you still get an error

P0W3, 2023-01-22:
Okay. After I used the 'make' command, it worked, but there is an error that appeared from MSYS.

```
______ MSYS ~
$ cd C:\priiloader

______ MSYS /c/priiloader
$ make
make -C loader
make[1]: Entering directory '/c/priiloader/loader'
-----------------------------------------------
compiling the loader...
-----------------------------------------------
loader.c
_stack.s
_startup.s
make[1]: Leaving directory '/c/priiloader/loader'
make -C ./tools/bootloader
make[1]: Entering directory '/c/priiloader/tools/bootloader'
copying loader...
building bootloader...
make[1]: g++: No such file or directory
make[1]: *** [Makefile:24: all] Error 127
make[1]: Leaving directory '/c/priiloader/tools/bootloader'
make: *** [Makefile:3: all] Error 2

______ MSYS /c/priiloader
$

```

DacoTaco, 2023-01-22:
ah, youre missing a regular gcc build in msys. 
in msys run 'pacman -Q | grep gcc'
does it list gcc?

if not : pacman -Syu gcc

P0W3, 2023-01-22:
Okay. After I used the 'pacman -Q | grep gcc' command, it shows that I have the gcc build in MSYS.

When I used the 'pacman -Syu gcc' command, it shows something about the 'msys.db' errors.

```
____ MSYS ~
$ pacman -Q | grep gcc
gcc-libs 11.3.0-2

____ MSYS ~
$ pacman -Syu gcc
:: Synchronizing package databases...
 msys is up to date
 dkp-libs.db failed to download
error: failed retrieving file 'msys.db' from downloads.sourceforge.net : error setting certificate verify locations:  CAfile: /usr/ssl/certs/ca-bundle.crt CApath: none
error: failed retrieving file 'msys.db' from www2.futureware.at : error setting certificate verify locations:  CAfile: /usr/ssl/certs/ca-bundle.crt CApath: none
error: failed retrieving file 'msys.db' from mirror.yandex.ru : error setting certificate verify locations:  CAfile: /usr/ssl/certs/ca-bundle.crt CApath: none
error: failed retrieving file 'msys.db' from mirrors.tuna.tsinghua.edu.cn : error setting certificate verify locations:  CAfile: /usr/ssl/certs/ca-bundle.crt CApath: none
error: failed retrieving file 'dkp-libs.db' from downloads.devkitpro.org : error setting certificate verify locations:  CAfile: /usr/ssl/certs/ca-bundle.crt CApath: none
error: failed to synchronize all databases (download library error)

____ MSYS ~
$

```

DacoTaco, 2023-01-22:
that seems to be a connection issue, it might mean your ISP is doing some parental control or your IP/location has somebody that has been abusing the devkitpro's servers and has been banned because of it.

its not safe to post your ip here, but i could ask wintermute to check. send me a msg with your ip on discord : DacoTaco#7948

DacoTaco, 2023-01-22:
did you do anything special stuff to your msys/devkitpro toolkit installation ? seems some stuff are broken inside of it :/

P0W3, 2023-01-22:
I have not done anything special to MSYS or the devkitPro toolkit installation.

AndrewPiroli, 2023-01-22:
pacman can't find your root certificate store. Try running `update-ca-certificates` and `update-ca-trust extract`. You may also need to force reinstall the ca-certifcates package from pacman or [manually reinstall it ](https://packages.msys2.org/package/ca-certificates?repo=msys&variant=x86_64) if you can't convince pacman to connect to HTTPS at all.

urherenow, 2023-01-22:
Looks more like his person hasn't installed devkitpro correctly at all. Start with "customizing existing pacman install" here:  https://devkitpro.org/wiki/devkitPro_pacman

And of course:  pacman -S wii-dev.

P0W3, 2023-01-22:
For me, devkitPro worked well with pacman a few years ago, but then the pacman installation error happened this year.

P0W3, 2023-01-23:
Okay. I used 'update-ca-certificates' and 'update-ca-trust extract', and it worked well. I then used 'pacman -Syu'. For everything else, this is what I have done so far.
[MSYS2_Progress.txt](https://github.com/DacoTaco/priiloader/files/10475947/MSYS2_Progress.txt)

AndrewPiroli, 2023-01-23:
So now that pacman is working, were you able to get a working host gcc/g++ (I'm not reading all of that)?

If so, try building priiloader again. Start by running `make clean` first to clean up anything weird from the failed build. Then run `make`

DacoTaco, 2023-01-23:
i have read most of it and wtf have you been installing.
msys stuff? mingw INSIDE msys?

urherenow, 2023-01-23:
The latest (msys2, actually) comes with mingw32 and mingw64 by default now…

I personally prefer ubuntu in a VM though. Unless I need a Windows executable.

DacoTaco, 2023-01-23:
ive personally moved to wsl hehe

P0W3, 2023-01-23:
Using MSYS towards the Priiloader folder, I used 'make clean' and then 'make'.

This is how it came out.

```
MSYS ~
$ cd C:\priiloader

MSYS /c/priiloader
$ make clean
make -C ./tools/bootloader clean
make[1]: Entering directory '/c/priiloader/tools/bootloader'
cleaning...
make[1]: Leaving directory '/c/priiloader/tools/bootloader'
make -C ./tools/OpenDolBoot clean
make[1]: Entering directory '/c/priiloader/tools/OpenDolBoot'
cleaning...
make[1]: Leaving directory '/c/priiloader/tools/OpenDolBoot'
make -C loader clean
make[1]: Entering directory '/c/priiloader/loader'
clean ...
make[1]: Leaving directory '/c/priiloader/loader'
make -C priiloader clean
make[1]: Entering directory '/c/priiloader/priiloader'
clean ...
make[1]: Leaving directory '/c/priiloader/priiloader'
make -C Installer clean
make[1]: Entering directory '/c/priiloader/Installer'
clean ...
make[1]: Leaving directory '/c/priiloader/Installer'

MSYS /c/priiloader
$ make
make -C loader
make[1]: Entering directory '/c/priiloader/loader'
-----------------------------------------------
compiling the loader...
-----------------------------------------------
loader.c
_stack.s
_startup.s
make[1]: Leaving directory '/c/priiloader/loader'
make -C ./tools/bootloader
make[1]: Entering directory '/c/priiloader/tools/bootloader'
copying loader...
building bootloader...
make[1]: Leaving directory '/c/priiloader/tools/bootloader'
make -C ./tools/OpenDolBoot
make[1]: Entering directory '/c/priiloader/tools/OpenDolBoot'
building OpenDolBoot...
make[1]: Leaving directory '/c/priiloader/tools/OpenDolBoot'
make -C priiloader
make[1]: Entering directory '/c/priiloader/priiloader'

!!!!!!WARNING!!!!!!!
--------------------
NOT BUILDING FROM GIT CHECKOUT. REV NUMBERS WILL NOT MAKE ANY SENSE

/bin/sh: -c: line 1: unexpected EOF while looking for matching `''
make[1]: *** [Makefile:162: pre_build] Error 2
make[1]: Leaving directory '/c/priiloader/priiloader'
make: *** [Makefile:5: all] Error 2

MSYS /c/priiloader
$

```

Here is a bottom message. - "Not building from Git Checkout."? Also, "Rev numbers will not make any sense."?

What specific error am I encountering?

urherenow, 2023-01-23:
It’s not being recognized as a cloned repo. Did you download the .zip? If so, you're doing it wrong. I wouldn’t clone into a regular drive anyway. 

Also, this issue has already gone on far too long, because Github is not a support forum. This space id for reporting bugs, and for developers to collaborate. This is neither. 

You should close this and ask for help on gbatemp or discord.

DacoTaco, 2023-01-23:
> 

> It’s not being recognized as a cloned repo. Did you download the .zip? If so, you're doing it wrong. I wouldn’t clone into a regular drive anyway.

downloading the source in zip should be supported by the build. apparently it was broken but andy fixed it before i could haha.

in any case, this is closed now as everything should work now

PR #322 (2023-01-23): Makefile: Fix git rev fallback path for non-git checkouts
Looks like an extra quote got left behind when refactoring this in b0e74f0
Before:
```
make[1]: Entering directory '/home/apiroli/work/priiloader-me/priiloader'

!!!!!!WARNING!!!!!!!
--------------------
NOT BUILDING FROM GIT CHECKOUT. REV NUMBERS WILL NOT MAKE ANY SENSE

/bin/sh: 1: Syntax error: Unterminated quoted string
make[1]: *** [Makefile:162: pre_build] Error 2
```
After:
```
make[1]: Entering directory '/home/apiroli/work/priiloader-me/priiloader'

!!!!!!WARNING!!!!!!!
--------------------
NOT BUILDING FROM GIT CHECKOUT. REV N [... cut]