DacoTaco/priiloader #323: Region Free Everything option is not working.

soulfunkdjx, 2023-03-23:
Hi recently i was playing just fine my NTSC games on my PAL Wii.
Then i decided to update my Cios and the priiloader to the latest versions.
I did it succesfully. And from priiloader the settings were gone so i enabled again the "Region Free Everything" option but unfortunatelly the NTSC games are not running anymore. I try their PAL version are running ok.
Can you give me any solution?
Thanks in advance.

DacoTaco, 2023-03-23:
what system menu version are you using? have you tried updating the hacks_hash.ini from the one in the repository?

soulfunkdjx, 2023-03-27:
Hi i am using the Wii PAL 4.3E menu system. It was working great before i update the priiloader to the latest version (0.9.1).
Where is that repository? Any link?

closed 2023-03-27 by soulfunkdjx