DacoTaco/priiloader #324: Resolution not properly working with GCVideo modification

SwareJonge, 2023-03-25:
**Describe the bug**
Currently when launching priiloader with a GCVideo mod (without resolution fix enabled) the video feed will be shifted all the way to the left or basically not get detected

**To Reproduce**
Install GCVideo mod into wii and load priiloader without resolution fix enabled

**Version**
0.10.0

**Expected behavior**
Priiloader would boot normally

**Additional context**
This might be an issue with GCVideo itself not being able to handle this custom resolution so this won't be an issue for most people, just wanted this to be documented and raise awareness

DacoTaco, 2023-03-26:
"resolution fix"? what fix are you talking about?

SwareJonge, 2023-03-26:
Resolution fix is an option in the GCVideo settings, I've experienced the same issue with first party titles so closing this issue

closed 2023-03-26 by SwareJonge