DacoTaco/priiloader #326: Add autoboot to channel

Lami-Lemmy, 2023-04-01:
Currently, priiloader allows you to autoboot to the hbc. Considering that priiloader already allows you to launch a specific channel, it would make sense to extend the autoboot option to let users autoboot to any channel on their Wii menu.

DacoTaco, 2023-04-02:
what channel would you set as autoboot? 
im asking this because right now i don't really see the point of the feature

Fan8099, 2023-04-02:
> what channel would you set as autoboot? 
> im asking this because right now i don't really see the point of the feature

They want to set the MKW-SP (Mario Kart Wii Service Pack) Channel as autoboot.
It is a mod for MKWii which includes many QoL features. Unlike other mods, it's not run from Riivolution, but from the Homebrew Channel. Optionally, a Channel of it can be installed onto the Wii Menu, allowing for quick access.

Installing the boot.dol file of MKW-SP does not work to autoboot into MKW-SP. Even just loading the boot.dol from Priiloader does not work. However, the Channel does work. This is why they want to autoboot into a Channel.

DacoTaco, 2023-04-02:
can the dol be shared?
it should work, and i think *that* is the real issue. autoboot of channels seems like a nasty work around for that real issue

Fan8099, 2023-04-02:
I can't seem to upload the .dol directly, so here's the link for it: https://github.com/stblr/mkw-sp/releases/download/v0.1.9/mkw-sp-v0.1.9.zip

Lami-Lemmy, 2023-04-03:
> it should work, and i think *that* is the real issue. autoboot of channels seems like a nasty work around for that real issue

When installing a dol like this can it be loaded without the SD card inserted?

DacoTaco, 2023-04-03:
i tried launching the dol from HBC and got a black screen. are you sure that dol is ok?

Fan8099, 2023-04-03:
Interesting. Now that you mention it, I get also get a black screen. Do you get it right away, or when MKWii should've loaded? I got it when everything was done MKWii should've booted up.
It used to work before. Now, not even loading it from the Channel works.

DacoTaco, 2023-04-03:
instant black screen if the disc is not inserted, black screen after some init otherwise. 
loaded through priiloader i get video artifacts, which probably means they do some weird video stuff wrong ...

Fan8099, 2023-04-03:
Apparently, having a LAN Adapter inserted speeds up loading into MKWii?!?

DacoTaco, 2023-04-03:
> When installing a dol like this can it be loaded without the SD card inserted?
Yes, as it is saved on the wii itself

> Apparently, having a LAN Adapter inserted speeds up loading into MKWii?!? Also, with it inserted, I get the weird video artifacts again.
this proves to me they are doing some weird video stuff that isn't right. i have a few ideas to make the artifacting when loading from priiloader to go away but they really need to get that checked imo...

closed 2023-04-03 by DacoTaco