DacoTaco/priiloader #328: 480 fix not working on revisions newer than

hiddejager, 2023-04-11:
The 480 fix in Priiloader does not fix the 480p problem on Wii model revisions newer than rev01.

To reproduce the behavior: 

240p testsuite -> enable 480p 1:1 scale in the option menu -> access the checkerboard menu
Without a OSSC or Retrotink5x and optimal sampling/timings enabled you might not see it. 
This is how it supposed to look like when 480p fix is applied correctly with rev01

Version: 0.9.1

Expected behavior: The 480p fix fixing the 480p problem. 

https://twitter.com/wobblingp/status/1645574502645112832?s=46&t=nlwbpxFYJTcoffO1IbX_eA

[image]

[image]

DacoTaco, 2023-04-12:
and this is in system menu or...?

hiddejager, 2023-04-12:
> and this is in system menu or...?

I believe both games and system menu. @Wobblingpixels on Twitter should have all the details

DacoTaco, 2023-04-12:
they are free to send in the fix for SM then...