DacoTaco/priiloader #330: Proper GameCube Mode Shutdown

Snivy102, 2023-04-20:
I was wondering if it's possible for the Wii console to shutdown fully after pressing the power button during GameCube game playback from a disc? It looks like the Wii just restarts, which is common for some GameCube loaders. (Note: Nintendont powers down properly)

DacoTaco, 2023-04-21:
it normally already should power down completely.
what are your shutdown settings in priiloader?

Snivy102, 2023-04-21:
[image]

DacoTaco, 2023-05-01:
bug confirmed. boot state needs to be invalidated before booting BC

Snivy102, 2023-05-01:
Is there anything I need to do?

DacoTaco, 2023-05-01:
nah, i need to do some investigation and reverse engineering on SM and see what needs to be done

DacoTaco, 2023-05-03:
do you happen to know of any other gamecube game loaders besides nintendon't that don't have this issue?

DacoTaco, 2023-05-03:
problem has been solved latest main branch. was an issue of closing the nand handle before setting the bootstate

closed 2023-05-03 by DacoTaco