DacoTaco/priiloader #331: Wii Discs Sometimes Don't Boot

Snivy102, 2023-04-21:
**Describe the bug**
Disc drive starts up, and just keeps spinning. The screen either gets stuck on "Loading DVD... -game name-" or a green corrupted screen. This only happens with Wii titles, and the type of game used doesn't matter.

**To Reproduce**
Steps to reproduce the behavior:
Hold Reset when powering on to go to Priiloader. Select the option to start the disc. This problem happens randomly. I could not find a pattern. I tried starting without the disc already in the Wii, tried with and without storage devices also in the Wii, etc, but no pattern found. 

**Version**
[0.10.0-beta2](https://github.com/DacoTaco/priiloader/releases/tag/0.10.0-beta2)
NTSC Wii (RVL-001) + NTSC Wii Discs

**Expected behavior**
Boots the game after the disc spins for a bit.

**Screenshots**
[image]
[image]

**Additional context**
GameCube discs do not experience this issue.

DacoTaco, 2023-04-21:
are there any games in specific that this happens with more often?
i see smash bros in your screenshot, but is that one of them?

Snivy102, 2023-04-21:
I only tested two games. Smash Brawl and Mario Kart Wii. The issue occurred randomly with both of those games, so I assumed it would happen to others as well. I can test more Wii games if needed.

Maybe there's something wrong with my Wii since it also has that issue where it's not shutting down fully with the GameCube disc launcher. Maybe I could try wiping the cIOSs on it.

Magicalus, 2023-06-29:
This just happened to me on Mario Kart Wii. Other games seem to be fine, and I don't have Smash to test.

Magicalus, 2023-06-29:
After further testing: putting the disc in many times consecutively, it worked perfectly. However, if I go into Priiloader, boot the Homebrew Channel, and then exit back to Priiloader (after having set it as my Return to,) the game freezes. This is consistent, and always works. However, if I reboot the Wii before launching the disc, it launches successfully.

Additionally, although I have the "Light slot on error" setting enabled, when it freezes the light doesn't turn on.

DacoTaco, 2024-08-02:
i tried the method @Magicalus, but it worked just fine. is this on a wii or WiiU(vwii) ?

Magicalus, 2024-08-26:
> i tried the method @Magicalus, but it worked just fine. is this on a wii or WiiU(vwii) ?

Twas a Wii, but it turns out the drive of my wii was dying a quite horrible death.

DacoTaco, 2024-12-25:
this should probably be retested with the current code in main. i wasn't able to reproduce this even before the video changes were made though :(