DacoTaco/priiloader #334: Wii gamepad sync problem

carlito666, 2023-05-22:
Wii stopped syncying with my controller. 
[A clear and concise description of what the bug is.]

Priiloader_v0_10b2
[what version is run]

DacoTaco, 2023-05-22:
in priiloader or in system menu/official software?

carlito666, 2023-05-22:
In both places. The controller is Wii plus. 

Is it possible to downgrade to the previous priiloader version through Priiloader menu?

I use a keyboard to navigate through, but it works within Priiloader only.

DacoTaco, 2023-05-23:
it is normal that you can't sync with the red button inside priiloader (or any homebrew). 
however, priiloader should not have any effect on syncing in other official software like system menu.

i suggest to try the wiimote on another wii and sync it using the red buttons from within system menu.

closed 2023-05-23 by DacoTaco

carlito666, 2023-05-23:
Well, thanks.