DacoTaco/priiloader #337: Autobooting to BootMii IOS causes a sort of loop

Naim2000, 2023-07-07:
**Describe the bug**
When turning on the Wii from Standby while Priiloader is set to autoboot BootMii as IOS, selecting the Wii menu will launch BootMii once again, and selecting the Homebrew Channel will freeze the Wii.

**To Reproduce**
Steps to reproduce the behavior:
* Go to WiiConnect24 in Wii settings and ensure Standby Connection is enabled.
* Restart the Wii and enter the Priiloader menu.
* Go to Settings and set AutoBoot to BootMii IOS.
  - Ensure `Ignore Standby` is off.
* Go back and select System menu.
* Press the power button and you will enter Standby.
* Turn the Wii back on, and you will enter BootMii.
* Press reset to select the Wii menu. You will enter BootMii once again.
* Press power once and reset once to select HBC. Your Wii is stuck on a black screen.

**Version**
Priiloader v0.9.1 (r0x092658e7)
System menu v513, IOS80

**Expected behavior**
The options in the BootMii menu would launch their respective applications.

**Screenshots**
[If applicable, add screenshots to help explain your problem.]

**Additional context**
Add any other context about the problem here.

DacoTaco, 2023-07-07:
sounds normal to me.
wii does a boot , which boots priiloader, priiloader boots bootmii, bootmii boots System menu, which loads priiloader, which loads bootmii etc etc.

the bootmii autoboot is to be used with things that setup the wii in a certain way like Emunand or something.

the only thing i don't understand is why booting HBC causes a freeze. this is booting HBC in bootmii?

DacoTaco, 2023-07-13:
the return to option is not triggered, as bootmii/boot2 sets the system up like it is a cold boot. there is nothing priiloader can do but follow the boot rules and launch autoboot settings.

like i said though, that options was meant for emunand/sneek/whatever and not actual bootmii.

as for the HBC crash, i want to investigate if i can reproduce it, but my money is on it being a bootmii/boot2/HBC issue

DacoTaco, 2023-07-16:
unable to reproduce the HBC bug. seems to be something specific to the case and/or bootmii/hbc

closed 2023-07-16 by DacoTaco