DacoTaco/priiloader #341: Channel forecast is no longer updated after installing Priiloader on wiiu

AymericKing, 2023-07-21:
**Describe the bug**
after installing Priiloader on wiiu, the Forecast Channel no longer works. The time date that displays when it last updates the weather remains stuck at 07/20 20:01 CEST. The only way to update it is to go to priiloader, go to system hack and then update it which is painful because it takes time and requires an sd card each time the console is reboot
**To Reproduce**
Steps to reproduce the behavior:
I followed the Wiiconnect24 tutorial https://youtu.be/DX8Nn40oQdY

**Version**
[Priiloader 0.10.0 RC3 : Version Kappa for Wiiu

**Screenshots**
[
[image]
[image]

larsenv, 2023-07-21:
You need to install Wii version of Forecast Channel with the patcher, not the vWii one.

impeeza, 2023-07-21:
https://gbatemp.net/threads/riiconnect24-patcher-brings-back-the-weather-channel-to-the-vwii.580716/

DacoTaco, 2023-07-21:
> You need to install Wii version of Forecast Channel with the patcher, not the vWii one.

what

Ingunar, 2023-07-22:
If you used the current patcher you need to use the Wii version of Forecast Channel, not the vWii version. 
RiiConnect24 patcher and [wii.guide](https://wii.guide/) are still not updated to match the Priiloader changes and will be updated soon, I hope.

This goes for both Forecast and News channels:
- You need to use The Wii channel version, not the vWii one
- The vWii one is set to never expire
- The Wii one will expire after about an hour of console inactivity (Being off)
- The Wii one and the vWii one will update after going idle for about 30 - 60min

....Also this is not an issue related to Priiloader...

DacoTaco, 2023-07-22:
i would like to keep this open for a while until @larsenv responds.
because i fail to see why they don't update the vwii instead of having this terrible system in which the *user* needs to install a channel that was not designed for the system its running on

larsenv, 2023-07-23:
@DacoTaco You can close it.

DacoTaco, 2023-07-23:
@larsenv : you didn't answer my question though. why isn't the patcher being updated to fix this change of the vwii channel?

larsenv, 2023-07-24:
The dude which maintains the patcher is currently busy and is currently travelling on another country and he didn't bring his computer. @DacoTaco

DacoTaco, 2023-07-24:
then i would prefer you say that, and not recommend people to do hacky, shitty bypasses. what you do on your repo's is your own but on my repo's i do not tolerate that stuff. specially since vwii was hacked long ago, and this issue was known for *at least* 2 years.

i have a level of quality control i'd like to keep, thanks.

closed 2023-07-24 by DacoTaco