DacoTaco/priiloader #342: Wrong alignment message when setting up "Return To" option

impeeza, 2023-07-21:
Using Priiloader 0.10.0 RC3, when you on the "Settings" menu, on the option "Return to" when you select the new item "Installed file", the alignment is off:

[image]

Is only Aesthetic and do not affect functionalism.

Seems to be the line 955 on the file \src\priiloader\source\main.cpp:

```
				case RETURNTO_FILE:
					PrintFormat( cur_off==1, 0, 128,    "             Return to:    Installed file   ");
```

And should be:

```
				case RETURNTO_FILE:
					PrintFormat( cur_off==1, 0, 128,    "             Return to:          Installed file  ");
```

Hard to show the 6 spaces before the "Installed"

Ingunar, 2023-07-21:
Yes, can confirm that this is the case.
Also `Installed File` instead of `Installed file` (For consistency)

DacoTaco, 2023-07-21:
thanks for reporting. this will be fixed on the next release

closed 2023-07-21 by DacoTaco