DacoTaco/priiloader #345: "Disable System Menu Background Music" doesn't work on 4.3E

jornmann, 2023-08-10:
**Describe the bug**
Enabling the Disable System Menu Background Music system menu hack doesn't disable the music.

**To Reproduce**
Steps to reproduce the behavior:
1. Own a Wii running SysMenu version 4.3E
2. Install Priiloader 0.10.0 RC3
3. Enable the "Disable System Menu Background Music" system menu hack
4. Go back to the Wii Menu

**Version**
0.10.0 RC3 on System Menu 4.3E

**Expected behavior**
The music not being there

DacoTaco, 2023-08-10:
@Ingunar : have you seen this one? 

from the looks of your document, it should have been disabled?
( https://hackmd.io/@Ingunar/PriiloaderTesting )

DacoTaco, 2023-08-10:
seems like its broke on 4.3E then :D

DacoTaco, 2023-08-10:
@jornmann : is this a wii, wii mini, vwii,...?

jornmann, 2023-08-11:
regular Wii

jornmann, 2023-08-11:
the hacks_hash you supplied worked perfectly! thank you :)

closed 2023-08-11 by jornmann