DacoTaco/priiloader #348: Priiloader with USB Loader GX - Graphical Issue

EchoedMike, 2023-08-20:
**Describe the bug**
When Priiloader is installed, the game tiles in USB Loader GX change appearance. The game tiles are no longer shaped like a Wii tile (corners are not rounded).  When selecting a game, the options and play buttons are invisible but still work if you click in the same location. 

**To Reproduce**
Steps to reproduce the behavior:
Installed Priiloader (no settings changed), Ran USB Loader GX

**Harware**
Wii U: wup 101(02)

**Version**
Priiloader: 0.10.0 RC3
USB Loader GX: 3.0 r1281
HBC: 1.1.3

**Additional Info**
Uninstalling Priiloader returns everything back to normal. 

**Screenshots**

Without Priiloader
[image]
[image]

With Priiloader
[image]
[image]

DacoTaco, 2023-08-20:
this sounds like a usbloader GX issue, and not priiloader. something for @wiidev, not me

closed 2023-08-20 by DacoTaco

Jacoby6000, 2024-08-02:
Just a heads up, I have a fix for this here: https://gbatemp.net/threads/usb-loader-gx-on-vwii-with-priiloader-fix.658777/#post-10467813

And I have a more robust fix in the works which I'll probably port over to dolphin so that it can work in vwii mode as well

DacoTaco, 2024-08-02:
> Just a heads up, I have a fix for this here: https://gbatemp.net/threads/usb-loader-gx-on-vwii-with-priiloader-fix.658777/#post-10467813
> 
> And I have a more robust fix in the works which I'll probably port over to dolphin so that it can work in vwii mode as well

just a tip : don't use content index to load the data. if anything would change in the future its broken again. the best fix would be to look for the u8 archive instead of using content index..

but im glad to see it finally fixed...

Jacoby6000, 2024-08-02:
I have a commit in my local repo that I made last night to search by content id. I have a list of relevant content Ids for both wii and vWii and I search for that rather than index. I haven't tested it yet, so I didn't bother pushing it.

What do you mean by "look for the u8 archive"? I'm new here 😅

DacoTaco, 2024-08-02:
> I have a commit in my local repo that I made last night to search by content id. I have a list of relevant content Ids for both wii and vWii and I search for that rather than index. I haven't tested [...]

afaik usbloader needs the u8 archive of system menu to load in all the textures etc.
it would be better to loop over all content and check if it is a u8 archive and contains the data you need
for more info about the fileformat : https://wiibrew.org/wiki/U8_archive