DacoTaco/priiloader #349: System menu background music removed on Wii U vWii, but not original Wii with 0.10.0 RC3

lemmingfactory, 2023-08-30:
**Describe the bug**
Minor issue:  With 0.10.0 RC3, enable removal of system menu music setting works on Wii U vWii (no vWii menu music when enabled), but not on original Wii (Wii menu background music plays regardless of setting enabled/disabled).  

**To Reproduce**
Enable remove system menu background music setting on original Wii, yet music still plays in background.

**Version**
0.10.0 RC3

**Expected behavior**
Expected no Wii system menu music when remove background music option enabled.  

Wii system menu music is removed, however, when option enabled in version 0.9.1, so I reverted Wii back to version 0.9.1, which works as expected.

DacoTaco, 2023-08-30:
what hacks file did you use? the release or the one fixed by commit 84095abcc0e1c4aaaf2a8441214c1c63b9d070e5 ?

DacoTaco, 2023-08-30:
try the one from [the commit i mentioned above](https://raw.githubusercontent.com/DacoTaco/priiloader/84095abcc0e1c4aaaf2a8441214c1c63b9d070e5/src/priiloader/hacks_hash.ini)

lemmingfactory, 2023-08-30:
The new hacks_hash.ini fixed the issue.  Thanks!  Perhaps you could put that new .ini file in the 0.10.0 download?  Nice to have the ability to customize vWii, especially removing music and health & safety screen.

closed 2023-08-30 by lemmingfactory