DacoTaco/priiloader #351: Error autobooting to System Menu with IOS249 (d2x cIOS)

GabubuAvailable, 2023-09-09:
**Describe the bug**
When I set to use IOS249 for System Menu, which is a d2x cIOS, then try to load it from Priiloader, I manage to get the following error message on screen: _see the screenshots_

**To Reproduce**
Steps to reproduce the behavior:
1. Open Priiloader
2. Go to settings then disable the `Use System Menu IOS` and set the `IOS to use for SM` to `249`
3. Save the settings then go back to the main menu
4. Select `System Menu` and "voila" error

**Version**
[0.10.0](https://github.com/DacoTaco/priiloader/releases/tag/0.10.0)

**Expected behavior**
Booting to System Menu without any issue.

**Screenshots**
Error showing on screen:
[image]

Settings screen (just the bottom page):
[image]

**Additional context**
The log: [prii.log](https://github.com/DacoTaco/priiloader/files/12566080/prii.log)

DacoTaco, 2023-09-09:
does that cios have ES_Identify(DIVerify) patched? the ES_Identify call failed, which is required if you want to start SM with an IOS besides the what is loaded when priiloader start.

could you try any other ios for a second?

GabubuAvailable, 2023-09-09:
> does that cios have ES_Identify(DIVerify) patched? the ES_Identify call failed, which is required if you want to start SM with an IOS besides the what is loaded when priiloader start.

I am not sure exactly how to check that. But I can say that it worked when I used [Priiloader 0.9.1](https://github.com/DacoTaco/priiloader/releases/tag/0.9.1).

> could you try any other ios for a second?

cIOS 249, 250 and 251, all resulting to the same error

DacoTaco, 2023-09-09:
with any other ios, i meant any *not cios* ios :)

GabubuAvailable, 2023-09-09:
It happens with every IOS I tried.
_Except for Stub IOS' where it actually tells me it's a Stub_

I also forgot to send my SysCheck (if you are interested): [SysCheck.csv](https://github.com/DacoTaco/priiloader/files/12566143/SysCheck.csv)

DacoTaco, 2023-09-09:
i'll have to test this on my own wii later. this used to work though...

Senseless-Creature, 2023-09-09:
Try to run SM with IOS60/70/80 😉

DacoTaco, 2023-09-09:
huh, thats odd. when run from HBC (just the dol, not installed) it works, but when installed it doesn't. i will have to dig deeper into this...

GabubuAvailable, 2023-09-09:
> Try to run SM with IOS60/70/80 😉

Those works. But I don't want to use them lol

GabubuAvailable, 2023-09-09:
> huh, thats odd. when run from HBC (just the dol, not installed) it works, but when installed it doesn't. i will have to dig deeper into this...

Take your time! No need to rush :)

DacoTaco, 2023-09-09:
> Those works. But I don't want to use them lol
Thats not what you said earlier ... :)
> It happens with every IOS I tried.
> Except for Stub IOS' where it actually tells me it's a Stub

anyway, i need to do some testing and stuff and i'll see where the problem lies

GabubuAvailable, 2023-09-09:
> Thats not what you said earlier ... :)
To be honest, I tested by randomly choosing an IOS, not all. Sorry ;-;

Zurdonx, 2023-09-09:
I can confirm that this ES_DiVerify problem is also happening on Wii U (vWii). I'm unable to launch the System Menu using any cIOS (d2x). Also, when I selected IOS80, it first gave me a distorted "System Files are Corrupted" error, but when I tried again it worked.

DacoTaco, 2023-09-09:
ye, that distorted message is part of the ES_DiVerify call not working/being processed correctly.

MasterKelvinVIP, 2024-02-13:
Hey is there any update to this? Would really like to install Priiloader onto my vWii!

TheTechRobo, 2024-02-13:
You can, can't you? You just can't select any cIOS to boot from in Priiloader's settings.

MasterKelvinVIP, 2024-02-13:
Well, it is installed but I would like to boot into a specific cIOS to get fakemote (d2x) up and running. Unless there is another way to get it running? Genuinely unsure

DacoTaco, 2024-04-01:
i had found some possible fixes (restarting es when patching & patching fakesign back in which is required because we changed the SM TMD) but other issues arose which sucks ass -_-

DacoTaco, 2024-04-01:
ok, so to explain this to those who understand : this issue exists in 2 parts.
one part is priiloader, which should also patch fakesign when doing an ES_Identify call as we altered the TMD on installation.
this should fix the corruption message in System menu people are getting.

however, there is a second part in libogc.
when IOS_Reload is called, it sometimes happens IOS hasn't fully init yet by the time it tries to reinit the IOS subsystems (ES) and therefor libogc is in a state where it can't do any ES calls until IOS is done loading fully and it opens /dev/es again.

normally, once libogc detects IPC is back online, it should be good to go but very very rarely you get a race condition in which IPC is up and running but /dev/es is not.

i've made a fix by trying to re-init the subsystems 3 times, and confirmed it working. but we are still discussing how we will integrate the fix and leave the code in a better state than it is right now because this is a nasty fix for a nasty bug tbh

DacoTaco, 2024-08-02:
im closing this as the fix was included in libogc's 2.8.0 release, making both issues fixed :)

closed 2024-08-02 by DacoTaco