DacoTaco/priiloader #355: Hack Enable Status - Index Bug

Ingunar, 2023-09-18:
**Describe the bug**
Changing the hack position or adding new hacks to the `hacks_hash.ini` will display the incorrect `enabled` or `disabled` status inside the `System Menu Hacks` section.

**To Reproduce**
If you add new hacks or move the hacks anywhere except the bottom of the `hacks_hash.ini` list, the Priiloader reads the hack status based on the Index position previously set. (Hacks not compatible with the system menu version count too)

**Version**
0.10.0

**Expected behavior**
The hacks `enabled` and `disabled` flags should match the previously set settings.

**Additional context**
This leads to full confusion about what is happening when saving new settings.

DacoTaco, 2023-09-18:
euh, adding a hack should show all hacks disabled iirc? 
or thats at least what the code *should* be doing :p

Ingunar, 2023-09-19:
Hmm trying to remember...

Does it check if you added/removed a hack with the same name?
Since we added and removed some min/max versions.

Ingunar, 2023-09-19:
[hacks_hash (1)](https://github.com/DacoTaco/priiloader/files/12664241/hacks_hash.1.txt)
[hacks_hash (2)](https://github.com/DacoTaco/priiloader/files/12664242/hacks_hash.2.txt)

From the Discord conversation:
> I replaced the first one via FTP, and used WiiU Priiloader Forwarder, and checked the enabled/disabled flags

DacoTaco, 2023-09-19:
if the amount of hacks stays the same it might be confused, but if the hacks amount changed they should all be disabled
just to clarify

Ingunar, 2023-09-20:
Okay, we can be sure about these two cases: 
- If you change the hack order
- If you add a new one in the place of the existing one. (Name and code can be changed)