DacoTaco/priiloader #360: Shutdown Bug

MeshariHat, 2023-10-24:
**Describe the bug**
If a game is launched then decided to turn off the console using the Wiimote it would take me back to the System menu

**To Reproduce**
Steps to reproduce the behavior:
1-Start the vWii
2-Start Priiloader from the Homebrew channel
3-Go to setting and set the first 3 options to:
•Autoboot: System Menu
•Return to: System Menu
•Shutdown to: Power Off
-Save setting and exit Priiloader
-Once you are in the System menu launch a game
-Turn off the console using the Wiimote
-Now it would take you to the System menu instead of turning off the console

**Version**
Priiloader 0.10.0 : Version Kappa

**Expected behavior**
Turn off or Shutdown the console

**Additional context**
i’m using Priiloader on Wii U’s vWii

Ingunar, 2023-10-24:
We are aware of this, and it is related to any title atm if the standby flag is enabled on vWii.
We still don't know how to fix it, and Priiloader can't fix that.

vWii can shut down from the System Menu since Priiloader will patch that bug with
**[Permanent vWii System Settings]** or **[Always enable WiiConnect24 on boot]**

(@GaryOderNichts 's fix in those hacks patches OSShutdown to ignore the `IPL.IDL` settings so it always shuts down and doesn't go to the standby, in vWii case returns to SM)

My guess is a some kind of shutdown fix could be created and added to the homebrew app.
(But that would not fix games or other titles)

TCAtrevor, 2025-06-22:
I've found that the `Ignore standby` setting solves this problem, despite the [FAQ](https://github.com/DacoTaco/priiloader/blob/master/docs/FAQ.md) saying it does nothing on vWii. (Although now when I tried to go back and disable that setting, it still shut down the console instead of going to the Wii menu, so I'm a bit confused...)