DacoTaco/priiloader #361: HBC Issues

Mac1Chip, 2023-11-16:
**Describe the bug**
Whenever I attempt to load Homebrew Channel from the Priiloader menu, it states "Error autobooting HBC, maybe title isn't installed?" Then, when I try booting into the System Menu after that error, it displays a garbled "System files are corrupted" message.

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to autoboot HBC
2. Attempt to autoboot SM

**Version**
0.10.0

**Expected behavior**
I expected Priiloader to autoboot into the System Menu with no issues.

**Additional context**
This originated after using 240p Test Suite's memory viewer.
After the bug occurs, all the titles in the "Launch Title" list become a string of question marks.
The system still boots perfectly fine, but it cannot boot HBC.

DacoTaco, 2023-11-16:
could you please note the IOS it says it is running before, and after attempting to boot HBC?
could you also go to boot title and tell me what title ID its showing for the homebrew channel?

DacoTaco, 2024-04-01:
closing due to missing information

closed 2024-04-01 by DacoTaco