DacoTaco/priiloader #362: Return to installed file not work

25minutes, 2024-01-06:
**Describe the bug**
[A clear and concise description of what the bug is.]
I install Priiloader v0.10.0 on Wii. Then install Nintendont Forwarder (https://github.com/jmlee337/Nintendont-Forwarder-for-Priiloader) from Priiloader menu. Configure Priiloader settings as below:
- Autoboot: Installed File
- Return to: Installed File

It autoboot into Nintendont correctly. However, when I quit game (with SELECT button from PS2 controller), it doesn't return to Nintendont. Instead it returns to Homebrew Channel.

I also try to set Return to: Autoboot, and result is the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure Priiloader as the description above.
2. Power Wii on, it opens Nintendont correctly.
3. Launch any game, then use the configured button on the USB controller to quit.
4. It cannot return to Nintendont.

**Version**
v0.10.0

**Expected behavior**
When the game quits, it should return to Nintendont. That way, I don't need use Wiimote.

**Screenshots**
N/A

**Additional context**
Add any other context about the problem here.

DacoTaco, 2024-01-06:
Hi, 
would you mind enabling the debug dumping in settings? ( "Dump Gecko output" ) 
after that, do a power cycle and let it boot nintendon't, boot a gamer and exit the game.
do another power cycle and boot into priiloader. disable the setting again and upload the prii.log that should now be on your sd or usb

also i by the sound of it it sounds like a nintendon't issue and not priiloader, as priiloader would only boot HBC if any of the settings was set to boot HBC but i just want to verify it with the log

25minutes, 2024-01-06:
Sure. Log attached. 
[prii.log](https://github.com/DacoTaco/priiloader/files/13849446/prii.log)

So Nitendont (or maybe the forwarder) determines what to launch when it quits?

DacoTaco, 2024-01-06:
by the looks of the log, it only booted SM? is this on a wiiu?

25minutes, 2024-01-06:
Sorry, my bad. I misunderstood. It's Wii. What I did was:
enabling the debug dumping -> boot System Menu (because cannot poweroff with power button) -> poweroff -> boot into Nintendont

This time my procedure is:
1. enabling the debug dumping and save
2. Press power 5 sec to force poweroff
3. Power on again. Nintendont lanches.
4. Start a game and quit.
5. It launches HBC
6. Poweroff console
7. Boot into Priiloader menu, turn off Dump Gecko output, save setting
8. Exit to main menu, select HBC
9. Launch Ftpii and download prii.log, but the log does seem right (see attachment)
[prii.log](https://github.com/DacoTaco/priiloader/files/13849630/prii.log)

DacoTaco, 2024-01-06:
yup, looking at the log im not seeing priiloader being booted 3 times, which means Nintendont is controlling the return to from a game. i'd suggest asking them for help/changes. nothing i can do here! :)

closed 2024-01-06 by DacoTaco