DacoTaco/priiloader #363: Error Autobooting HBC + System Files are Corrupted

DefeatOf13, 2024-01-29:
**Describe the bug**
Whenever i try to boot into the Homebrew Channel via Priiloader i get an error message saying "Error Autobooting HBC. Maybe title not installed?". I can't even boot into the system menu, if i try to do it another error shows up saying "The system file are corrupted, blah blah blah.". BootMii works fine though.

**To Reproduce**
1. Boot into Priiloader
2. Press "Homebrew Channel" and get the error "Error Autobooting HBC. Maybe title not installed?"
3. Press "System Menu" and get the "System files are corrupted" error

**Version**
- Priiloader 0.10.0
- HBC 1.1.2
- System Menu 4.3E (v514)

**Expected behavior**
Boot into HBC and sysmenu without any issues.

**Screenshots**
[image]
[image]

impeeza, 2024-01-29:
May be you have a soft bricked nand, what if you try to load / install a file utem and then load a basic homebrew from your SD card? By example syscheck

DefeatOf13, 2024-01-29:
> May be you have a soft bricked nand, what if you try to load / install a file utem and then load a basic homebrew from your SD card? By example syscheck

Ok, i did as you said. I'm able to load / install apps without any problems (and they boot too). I tried Syscheck ModMii edition and USB Loader GX

DacoTaco, 2024-01-29:
check the System menu IOS setting in priiloader. what are the settings?

DefeatOf13, 2024-01-29:
> check the System menu IOS setting in priiloader. what are your priiloader settings in general?

The "Use System Menu IOS" option is already checked.
Here's my settings: 
[image]

DacoTaco, 2024-01-29:
hmm, i too think you might have a softbrick. did you recently install any channels? 
also, try the following things : 
* disable autoboot and boot HBC when menu shows up, don't hold reset on boot. does it boot HBC?
* keep autoboot disabled, and go to boot channel, can you boot HBC from there? what HBC is installed?
* put autoboot back on System menu, and set use system menu ios to off. as IOS, select a CIOS (like 249). boot SM. does it boot SM?

DefeatOf13, 2024-01-29:
Update.

I went back to "Boot Title" and now all the titles installed onto the console are called "????????"

[image]

impeeza, 2024-01-29:
Yeah you have a corrupted NAND try, run ModWii https://modmii.github.io/ (https://gbatemp.net/threads/best-way-to-mod-any-wii-modmii-for-windows-official-support-thread.207126/) on your computer and reinstall everything, you can load the HBC installer from Priiloader.

DefeatOf13, 2024-01-29:
> Yeah you have a corrupted NAND try, run ModWii https://modmii.github.io/ (https://gbatemp.net/threads/best-way-to-mod-any-wii-modmii-for-windows-official-support-thread.207126/) on your computer and reinstall everything, you can load the HBC installer from Priiloader.

Like... What exactly do i need to reinstall?

DacoTaco, 2024-01-29:
> Yeah you have a corrupted NAND try, run ModWii https://modmii.github.io/ (https://gbatemp.net/threads/best-way-to-mod-any-wii-modmii-for-windows-official-support-thread.207126/) on your computer and reinstall everything, you can load the HBC installer from Priiloader.

wrong. priiloader is fucking with the loaded IOS and SM can't load certain files.

DefeatOf13, 2024-01-29:
What do you suggest to do? (always if there is something that can be done)

impeeza, 2024-01-29:
Oh, @DacoTaco a wrong priiloader like the vWii on the Wii?

so he could reinstall priiloader to try to fix the problem?

The priiloader installer can be loaded from priiloader?

DacoTaco, 2024-01-30:
i suspect this was an Priiloader setting issue together with invalid HBC installation or somehow it had a titleID that is unknown to Priiloader.

we will never know now though :)

closed 2024-01-30 by DacoTaco