DacoTaco/priiloader #364: Wii does not shut down in Gamecube mode

LightningTwo, 2024-02-22:
**Describe the bug**
Pressing the Power button while booted in a Gamecube game returns to the Priiloader menu. Then selecting 'System Menu' shuts down the Wii instead of loading the Wii System Menu.

**To Reproduce**
Priiloader settings:
Autoboot: Disabled
Return to: System Menu (or Priiloader)
Shutdown to: Power Off

Boot a Gamecube disc from Priiloader (Launch Disc). Press the Power button to shutdown the Wii. The Return To setting does not make a difference (System Menu or Priiloader).

**Version**
PAL Wii, IOS v80, System menu v514, Priiloader v0.10.0

**Expected behavior**
I would expect the Wii to completely shutdown. 

**Additional context**
I looked into the code a bit, and maybe found an explanation why the Wii shuts down after selecting System Menu, because apparently the SM handles the shutdown after Gamecube mode. But perhaps this piece of code is never reached and `BootMainSysMenu()` is never called. Perhaps some of the flags are reset during reboot? Setting Autoboot to System Menu actually shuts down the Wii correctly.
```
case TYPE_UNKNOWN: //255 or -1, only seen when shutting down from MIOS or booting dol from HBC. it is actually an invalid value	
	if (flags.flags == FLAGS_STARTGCGAME && GcShutdownFlag != 0) //&& temp.discstate != 2)
	{
		//if the flag is 0x82 & the GcShutdown flag is set, its probably shutdown from mios. in that case system menu 
		//will handle it perfectly (and i quote from SM's OSreport : "Shutdown system from GC!")
		//it seems to reboot into bootstate 5. but its safer to let SM handle it
		gprintf("255 : System Menu");
		BootMainSysMenu();
	}
```
Thank you for your help.

Ingunar, 2024-02-23:
Did the bug return? https://github.com/DacoTaco/priiloader/issues/330

DacoTaco, 2024-08-02:
@LightningTwo : do you have bootmii@boot2 installed? is it set to run on startup?

LightningTwo, 2024-08-02:
@DacoTaco I have bootmii installed as IOS, my Wii unfortunately didn't support install as boot2. I don't know how to run it at startup. The first screen I see when I turn on my Wii is the priiloader screen, so I can easily boot Gamecube games without using a Wii remote. An autoboot setting to boot any disc would be awesome by the way.

DacoTaco, 2024-08-02:
np, just investigating the issue atm. bootmii@boot2 can change the bootstate a bit so thats why i was asking.
i have confirmed it not working, but im not sure why it broke again...
it looks like the bootstate flags are not being set correctly or something.

DacoTaco, 2024-08-02:
problem has been found, it randomly could fail loading the wii state file because the variable it was loading them into wasn't aligned and IOS said fuck you to the read request haha.

its been fixed and will be included in the next version that is released

closed 2024-08-02 by DacoTaco