DacoTaco/priiloader #365: Installing priiloader on my vwii added graphical glitches to usb loader gx

PalmTamino, 2024-02-25:
**Describe the bug**
Icons that should be squircle were rectangle, no start button after clicking on app

**To Reproduce**
try usb loader gx

**Version**
0.10

**Expected behavior**
no glitches

**Screenshots**
[image]
[image]

DacoTaco, 2024-02-25:
this is, afaik, a bug with USBloader GX. they need to fix their System menu theme loading code on vwii. we have provided them the fix more than 6 months ago, they should've fixed it by now...

closed 2024-02-25 by DacoTaco

PalmTamino, 2024-02-25:
> this is, afaik, a bug with USBloader GX. they need to fix their System menu theme loading code on vwii. we have provided them the fix more than 6 months ago, they should've fixed it by now...

maybe i’m using the wrong usb loader gx then, some old version? There are a lot on github, could you link me the one u suggest?

Many thanks 🙏

DacoTaco, 2024-02-25:
i don't know which one is the official one, but as far as i know they haven't released an updated version yet to fix this bug -_-

eku, 2024-02-26:
It hasn't been fixed so far.

PalmTamino, 2024-02-26:
> It hasn't been fixed so far.

yes, i think usb loader gx is pretty inactive in development, i hope there will even be a new update. If a dev could compile it with the fix themselves, that would be awesome (and fork it on github)