DacoTaco/priiloader #366: [feature] Add an option to unmount wiiu hdd

dracula99gr, 2024-03-08:
**Describe the bug**
Having connected 2 hdds one for WiiU (wiiu formatted) and other for Wii (fat32 formatted) makes loaders to crash or not work using cios 249 or 250 (I installed latest vWii v11 cioses as stated in the guide). However using 251 on usb1 port does the trick but certain games don't work with that. (on 251 they blackscreen and if you choose 249 or 250 they exit to vWii system menu) Wiixplorer always works (whatever port  and cios you use for the hdds) and never crashes. Since this issue always appears if both hdds are connected and the WiiU one is mounted from the WiiU side, it would be cool if Priiloader unmounted a hdd (specified by port or detected automatically that is a wiiu one) 

**To Reproduce**
Steps to reproduce the behavior:
0) having a wiiu hdd connected on port 0 and wii hdd connected on port 1 (or the other way around) 
1) Launch wiiu 
2) Make sure that wiiu has loaded/mounted the hdd
3) Booting vWii
4) Opening usbloadergx or WiiFlow lite using cios 249 or cios 250
5) congrats u get a blackscreen or a crash on refreshing devices 
**Version**
priiloader: 0.10.0
usbloadergx: 3.0 r1281
WiiFlow lite: 5.5.4

**Expected behavior**
No crashing while loading the wii disks regardless of cios or port when having both disks untouched

DacoTaco, 2024-03-10:
Hi,
Im sorry to say but in the design of the wii or vwii, the system is not running an OS that is managing the drives and which are mounted and which are not. and priiloader is also not that.
if an application, outside of priiloader, has issues with drives it should be those apps that need fixing or managing that stuff.

i get the issue though, and its been a long discussed issues whether homebrew should, or should not, mount wiiu drives. but if this is causing issues with loaders, and not priiloader, i would report this to them.

closed 2024-03-10 by DacoTaco

dracula99gr, 2024-03-18:
Welp I don't know if they still develop their apps. Could that issue be a cios issue that would require patching/modifying there? 
Hmm is it possible for priiloader to disable specific usb ports? That could be another approach