DacoTaco/priiloader #368: Error autobooting file

mpzfm1, 2024-04-20:
**Describe the bug**
Error "autobooting file, try to reinstall"

**To Reproduce**
Steps to reproduce the behavior:

Clic on installed file option

**Version**
0.10.0

**Expected behavior**
I dont know, what is this option for???

**Screenshots**
[If applicable, add screenshots to help explain your problem.]

**Additional context**
Add any other context about the problem here.

DacoTaco, 2024-04-20:
this option is to boot the binary that is installed through the `Load/Install File` option. this is for when you stopped the autoboot to do something and want to launch your installed/autoboot file :)

closed 2024-04-20 by DacoTaco

mpzfm1, 2024-04-20:
Thanks very much, and what app do you recommend to have installed?

impeeza, 2024-04-20:
> Thanks very much, and what app do you recommend to have installed?

Any you like, I personnaly have a WiiFlow forwarder DOL installed, so each time I start my console Wii-flow run.