DacoTaco/priiloader #371: Deflicker Filter Disable and USB Loader GX

devlman127, 2024-06-10:
With USB Loader GX being able to disable Deflicker filters through it's Loader Settings, is it possible for a similar feature for installed or disc games to be implemented or is it not possible?

Ingunar, 2024-06-10:
**Theoretically possible** - Maybe, but it is not as easy or practical as you might think
(Advanced hacks like Wiimmfi eat up memory, and this one is even more complicated)

**Is it planned**  - No, since USB Loaders can already do it, and they can do so much more title patching
(Region-free play works a lot better with Loaders for example)

---

I saw your comment on evWii by the way, you need to patch the main dol of every title to disable the the filter.
IIRC, USBLGX uses DB of dols that need patching since some titles have multiple dols. 
(Like I said, a lot more complicated)
There should be a tutorial on how to do it somewhere on gbatemp.

closed 2024-06-10 by DacoTaco

devlman127, 2024-06-10:
Just wish this could've been done for the cleaner user experience of just using wii u menu injects or wii channels instead of having yet another menu for games, but I guess I can't force you to do so.

DacoTaco, 2024-06-10:
its mostly a technical limitation tbh. the wii memory is limited and System Menu & Games uses a lot of it in MEM1 and MEM2 is unreliable as it is used by both SM & Games at random.

devlman127, 2024-06-10:
[cafe2wii patcher](https://github.com/V10lator/c2w_patcher) shows that you can increase the clock speed in vwii mode, though I'm guessing accessing separate hardware is more difficult than simply increasing the clock speed?

Ingunar, 2024-06-10:
Like Daco said, technical limitations, so the user experience is limited by it. You would have to think within the scope of it.

My personal suggestion for you is to talk to USBLGX and  WFL Loader devs for better channel/title forwarder support.
(That would also be good for Wii NAND's health since less writing and more space.)

Also, you can ask WiiLink to provide pre-patched channels.
(Since WC24 channels are the only ones that kinda should be on Wii's NAND)

Regarding the c2w patcher, IIRC it was set to work only with WiiU VC titles since the clock is set from WiiU Mode.
You would need to contact its dev. I vaguely remember there being some problem when loading Wii Mode with a boosted clock...