DacoTaco/priiloader #372: Improve the load title warning screen

Quadraxis-v2, 2024-07-10:
I have used [wad2bin](https://github.com/DarkMatterCore/wad2bin), an application that generates some "bogus wads" that you must install in your Wii. Now when I select the "launch title" option in the Priiloader menu this screen shows up for every wad:

[image]

I'm not sure if this also happens for title contents that are moved to an SD card using the official method. This warning stays for 2-3 seconds for every title and they can't be skipped, which is annoying. Would it be possible to show them all at once, or press A to skip them or some other solution? Thank you

DacoTaco, 2024-07-10:
this is before it shows the list of installed titles, right?
do you know what title 00010001-57324750 is?

the error is saying that it failed to retrieve that title's info, but im not sure what state that title is, and if it is a bug in priiloader or the title is wrong :/

i'd rather figure that out first, before adding the skipping :)

Quadraxis-v2, 2024-07-10:
> this is before it shows the list of installed titles, right? do you know what title 00010001-57324750 is?

Yes, that's right. It's W2GP, which is Phoenix Wright: Ace Attorney Justice for All.

> 
> the error is saying that it failed to retrieve that title's info, but im not sure what state that title is, and if it is a bug in priiloader or the title is wrong :/
> 
> i'd rather figure that out first, before adding the skipping :)

I made a dump and I saw that the content and data folders inside /title are completely empty. There is no TMD, probably because I never launched it. The ticket is indeed inside /ticket though. The same is true for all the other titles that give errors, their directories are empty.

DacoTaco, 2024-07-10:
that seems wrong to say the least. no TMD means the title has no description and SM should crash if it tries to load that (afaik)
on top of that, no content is also weird.
is this title moved to SD or something?

Quadraxis-v2, 2024-07-10:
> that seems wrong to say the least. no TMD means the title has no description and SM should crash if it tries to load that (afaik) on top of that, no content is also weird as that means the title/channel has no data, no program, no banner, nothing.. is this title moved to SD or something?

Yes, that's the case, wad2bin takes a wad and places the contents inside the SD and just leaves a bogus wad (with only the ticket apparently, although their readme says this, but it must be inaccurate):

> Both ticket and TMD for each converted WAD package must be installed on the target Wii console in order for this to work. For this matter, the program generates a bogus WAD package at the provided output directory, using the <title_id>_bogus.wad naming convention. It can be used with regular WAD Managers to install both ticket and TMD if needed.

This way you can batch install a bunch of wads directly to the SD card. I made the test and launched that previous title, the TMD appeared afterwards. The SD only has a content.bin, which according to Wiibrew has an [embedded backup WAD](https://wiibrew.org/wiki/Content.bin#Embedded_backup_WAD), so it's possible that it's taken from there. As expected Priiloader stopped giving me the warning for that particular title, but it would be nice if I could hide or skip the warnings.

DacoTaco, 2024-07-10:
i understand, but the warning is serious and its a possible sign things are in a brick state.  i don't know how that tool made its stuff, and why only after the first boot it sets up the title correctly. the tmd *should* be on the wii and if it isn't the tool has a bug and is fucking with the wii's internal workings. i follow the title/sd logic and the tmd is needed to know the title's name/banner/icon/whatever.

as the problem is solved i will close this. that wad2bin tool is at fault here and a correct fix would be to fix that tool, not to have priiloader ignore those warnings by adding a skip button.

closed 2024-07-10 by DacoTaco