DacoTaco/priiloader #374: Title launcher does not list game channels

Quadraxis-v2, 2024-08-29:
**Describe the bug**
The title launcher doesn't show those titles considered game channels, i.e. those that are stored inside `/title/00010004/` 

**To Reproduce**
1. Install a game channel, such as the Mario Kart Wii Channel
2. Choose "Launch title" in Priiloader

**Version**
0.10

**Expected behavior**
The list should include the game channels installed

**Screenshots**
[image]
[image]
[image]

DacoTaco, 2024-08-29:
https://github.com/DacoTaco/priiloader/blob/master/src/priiloader/source/titles.cpp#L555

you seem to be right that these channels are indeed not listed.
the code seems to indicate these are hidden channels. aren't they?
i can easily list them by making also processing them in the code i linked above :)

Quadraxis-v2, 2024-08-29:
Yeah, I was looking at that code right now. Seems like an easy fix.

DacoTaco, 2024-08-30:
ES_GetTitles should work fine, as SM is just a nintendo SDK application and uses IOS/ES as well.
i made a version you can install to try and see if the channels show up correctly/boot : 
[boot.zip](https://github.com/user-attachments/files/16812823/boot.zip)

Ingunar, 2024-08-30:
`00010004` naming is bad even on WiiBrew, I wanted to change it to something like *Disc-exclusive channels and Disc Games that have them` or something like that.

When we checked that Photo Channel code in the System Menu we found confirmation that read rules for `00010003` and `00010006` titles exist, but we still don't know any channels with those IDs.

`00010007` is still unknown.

`00010008` are the hidden ones EULA, Region Select, and SPD.

Not sure if any other should be added, to the `Launch title` list.

Quadraxis-v2, 2024-08-30:
> ES_GetTitles should work fine, as SM is just a nintendo SDK application and uses IOS/ES as well. i made a version you can install to try and see if the channels show up correctly/boot : [boot.zip](https://github.com/user-attachments/files/16812823/boot.zip)

Yes, this works. By the way, if you take a look at the screenshots I posted, titles whose names are shorter than 8 characters get padded with "?". This stems from: https://github.com/DacoTaco/priiloader/blob/master/src/priiloader/source/titles.cpp#L531

DacoTaco, 2024-08-31:
@Ingunar : what you mentioned is familiar with what is in priiloader. any upper title id it doesnt know will be ignored anyway :)
@Quadraxis-v2 : good to hear it worked. ill have to change that code as well but by the looks of it the GetTitleName function should set the name correctly, it probably misses a \0 at the end, which i will fix once im home, but im on vacation haha :)

DacoTaco, 2024-09-02:
@Quadraxis-v2 : do you have the .app or title id of the title that has the invalid/short name so i can investigate the issue?

closed 2024-09-05 by DacoTaco