DacoTaco/priiloader #375: Seems My vWii Bricks When Using PriiLoader - No One's Been Able to Help Me Resolve This

iWockstar, 2024-09-16:
**Describe the bug**
After installing PriiLoader, many of my vWii apps began giving me black screens that are only able to be resolved by restarting my Wii U. Specifically, this happens to the "Return to Wii U Menu" WAD, the HBC, the Wii Options button when I turn it into the Wii system settings via PriiLoader, and it happens when attempting to exit USB Loader GX. It may happen with other apps, but I've only tested these apps and I also tested Wii Links restored Wii channels, but those all worked fine. To resolve this, I've tried decaffeinating lightly and aggressively, but neither fully solved the issue. Everything would seem to wrok well but then I'd reinstall PriiLoader and it wouldn't work. I even attempted to uninstall PriiLoader via the PriiLoader installer, turn off my Wii U, turn it back on and boot into the vWii and I'll get the same issues. It's only resolved if I run vWii Decaffeinator and don't reinstall PriiLoader after finished the process (which I want for brick protection and Wii Link channel revival support). I have not found another person with this issue. Maybe one person with a similar issue, but they were able to use vWii NAND Restorer to get it working again. The issue is, I added that program to my wiiu/apps folder and nothing shows up. I don't think it works on Aroma or possibly anymore at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Was installing Wii Link to revive the old Wii channels a few months ago and during my installation process I selected Return to Wii U Menu to proceed with the next step of the guide and got stuck on a black screen. Had to unplug to restart (the first step of the guide was to install PriiLoader, so I probably finished installing it and attempted to return to the Wii U Menu then experienced this for the first time) - I want to add, I had PriiLoader when I first hacked my vWii but, as you probably know, it causes visual issues within the start game screen of USB Loader GX so I uninstalled it, but I NEVER ran into any of these issues I'm describing or even other issues until I reinstalled PriiLoaderfor Wii Link.
2. The black screen issue Kept Occurring with the Return to Wii U Menu WAD and it was happening to USB Loader GX and the HBC as well
3. Was told to decaffeinate lightly
4. Once fully complete, the Return to Wii U Menu and HBC WAD worked and I was able to exit out of USB Loader GX without getting a black screen
5. Installed PriiLoader
6. Turned on settings stated in Wii Links guide (Fix NWC24iSetUniversalTime from the menu, enable Permanent vWii System Settings, Wii System Settings via Options Button, and Create message via Calendar button. I also chose to turn on Remove Deflicker and Region Free Everything)
7. I checked the vWii Options to see if the Wii settings were there, they were. I closed it to reconnect my Wii U to wifi via the Wii U settings
8. Once connected I opened the vWii Menu again, close the vWii Options button in the bottom left corner and my screen went black
9. Opened USB Loader GX, then tried to close it, but my screen went black
10. Same with the Return to Wii U Menu WAD and the HBC WAD
11. I then uninstalled PriiLoader via PriiLoader installer and restarted my Wii U
12. Checked to see if the issues still occurred; it did
13. Uninstalled USB Loader GX and restarted my Wii U, the issues still occurred
14. Decaffeinated on aggressive mode
15. Followed the vWii hacks guide to rehack my vWii and got to the install PriiLoader step
16. I installed it, exited to the HB Menu, from the HB Menu exited to the system menu, which brings you right into the PriiLoader settings
17. Chose System Menu, the first option, and the screen went black so I had to unplug my Wii U. Now I'm here

**Version**
v0.10.0 (r0x02b70b36)

**Expected behavior**
Access my vWiis apps, system menu features, WADs, and homebrew and have them open and close without my screen going black and me needing to hold the power button for 4 seconds or unplug the Wii U.

**Screenshots**
N/A

**Additional context**
If I need to delete files off my my Wii U's SD Card I'm down to do it. I'll delete every file related to the vWii if I need to, I just want to get this resolved. If it's possible another vWii homebrew is messing with it maybe I can try to get rid of that too. I remember using BootMii and some other stuff for my vWii (not exactly sure what it's just a faint memory and a slight possibility this is the issue) a while back so maybe all that caused it and if I wipe everything vWii related it will fix it. Not sure though.

GaryOderNichts, 2024-09-16:
What region is your console and what system version are you on?

iWockstar, 2024-09-16:
> What region is your console and what system version are you on?

USA and v5.5.6 U

iWockstar, 2024-09-16:
Well now I don't know if it is PriiLoader messing up my vWii, it might b something else or maybe PriiLoader broke my vWii really badly at first a few months ago because when attempting to rehack my vWii, I ran vWii Compat Installer, opened my vWii, and then the HBC, but then I got the black screen I was talking about without even redownloading PriiLoader

DacoTaco, 2024-09-16:
ok so, i don't think this is directly related to priiloader but can you test the following?
* decaf vwii so it all works again
* install priiloader and leave all settings & hacks on default, do not change them
* test a channel or usbloader gx to see if everything still works

if that works do the following : 
* boot into priiloader and enable the hacks
* boot into SM, launch HBC and test a channel again and see if it black screens

if that works, do what the wiilink guide says and try it again.

i have a feeling it has something to do with the guide and something getting fucked up :)

iWockstar, 2024-09-16:
Sounds good, I'll give that a try in once I'm back from class. I have two classes today so it'll be a few hours. Thanks!

iWockstar, 2024-09-16:
> * l priiloader a

I decaffinated again but right when it was finished I attempted to open my, unhacked, vWii and it worked. Then when choosing the return to Wii U Menu WAD, my screen went black. I think my vWii's just fried now somehow. Therefore, I'm not sure if it's due to PriiLoader even though post-installing it a few months ago is when I first started experiencing the issues.

Do you have any suggestions on what I can do to fix it? If it helps my actual Wii U seems to be working fine

iWockstar, 2024-09-16:
> ok so, i don't think this is directly related to priiloader but can you test the following?
>
> * decaf vwii so it all works again
> * install priiloader and leave all settings & hacks on default, do not [...]

I decaffinated again but right when it was finished I attempted to open my, unhacked, vWii and it worked. Then when choosing the return to Wii U Menu WAD, my screen went black. I think my vWii's just fried now somehow. Therefore, I'm not sure if it's due to PriiLoader even though post-installing it a few months ago is when I first started experiencing the issues.

Do you have any suggestions on what I can do to fix it? If it helps my actual Wii U seems to be working fine

DacoTaco, 2024-09-17:
@GaryOderNichts : any ideas?

Ingunar, 2024-09-17:
> I decaffinated again but right when it was finished I attempted to open my, unhacked, vWii and it worked. Then when choosing the return to Wii U Menu WAD, my screen went black. I think my vWii's [...]

Did you do full decaf last time?

Ingunar, 2024-09-17:
Do you have anything connected to your USB ports?

iWockstar, 2024-09-17:
> Do you have anything connected to your USB ports?

Yeah, 99% of the time I have a USB Hub that connects to my Mayflash GameCube Controller Adapter and the UGREEN LAN Cable Adapter in the front two. I have my HDD connect via Y-Cable in the back two.

Not for this reason, but I've tried it with only the usb hub/mayflash plugged in, with all three plugged in, the usb hub/mayflash and the Ethernet, and the usb hub/mayflash and the HDD. I have not tried without the mayflash and usb hub. I have no tried moving the mayflash to the back ports, I have not tried moving the HDD to the front ports.

I will say, the setup I run 99% of the time was working perfectly until I first began experiencing issues after installing Priiloader,Priiloader was most likely not the issue it was just when thd issues started occurring a few months ago

DacoTaco, 2024-09-17:
OH. try unplugging all usb devices and then going back. its possible a usb device is making priiloader hang

iWockstar, 2024-09-19:
> OH. try unplugging all usb devices and then going back. its possible a usb device is making priiloader hang

Sorry I'm late to this. By going back do you mean decaffeinating again? Or do you just mean turning off my Wii U, unplugging the USB devices, then go back into Priiloader.

Thanks

Ingunar, 2024-09-19:
> > OH. try unplugging all usb devices and then going back. its possible a usb device is making priiloader hang
> 
> Sorry I'm late to this. By going back do you mean decaffeinating again? Or do you just mean turning off my Wii U, unplugging the USB devices, then go back into Priiloader.
> 
> Thanks

Just disconnecting USB devices would be fine.

DacoTaco, 2024-09-20:
> Just disconnecting USB devices would be fine.

i'd say decaffeinating again (to be sure its not a vwii setting or channel that is fucking it up now), installing priiloader again, check if it hangs, and unplug all usb devices. see if it works after unplugging the devices.

iWockstar, 2024-09-20:
> > Just disconnecting USB devices would be fine.
> 
> 
> 
> i'd say decaffeinating again (to be sure its not a vwii setting or channel that is fucking it up now), installing priiloader again, check if it hangs, and unplug all usb devices. see if it works after unplugging the devices.

Sadly, it's doing the same thing. I think my vWii might be fried. (It works enough if I'm cool with resetting it after changing games at times. It does for sure do the black screen less often now, it just still does it meaning my vWii doesn't work how it should). I'm starting to think my best bet would be to use this Wii U/vWii as long as possible. Then whenever I get too frustrated with it giving me a black screen, buying a new Wii U and just transferring everything over from my current Wii U/vWii to the new Wii U/vWii. I'll probably also get a normal Wii too so I can get the GC memory card slots for accessories such as the Mario Party mic. I also feel like it's going to be more reliable for hacking.

Regarding this, let's say I get the new Wii U/vWii and/or a new Wii. Would you suggest I start from scratch? (Aka use a fresh SD card in the new console). If I start from scratch on the Wii/vWii should I also wipe the HDD that my Wii U uses? (The HDD literally only contains Wii U games, updates, and DLC. Nothing else)

I know little about tech so I just do not want to take the risk of transferring my current issues over to the new console if that's even a possibility. If formatting the HDD and SD is even 1% safer, I'll take that route.

Also if starting from scratch is the best method, please let me know if you recommend I buy a new SD card and a new HDD, just a new SD card, just a new HDD, or if I should just format them. I'm cool with any option. 

Thanks

DacoTaco, 2024-10-20:
im closing this as it does not look to be priiloader related tbh.
i was expecting a hack to be the one doing the damage, but thats unknown now.
what i do think, is that some wad or channel is fucking with this together with some hacks.

if you do things from scratch on a new wiiu/vwii try and test it along *every* step along the way until it breaks and make not of where it broke. 
if you know something, let us know and i might reopen this then :)

closed 2024-10-20 by DacoTaco