DacoTaco/priiloader #376: Launch Title starts NTSC channels at 480i on PAL Wii

pyorot, 2024-10-24:
**Describe the bug**
On a PAL Wii set to output 480p in system settings (and correctly outputting 480p on the system menu), launching an NTSC N64 channel via Priiloader Launch Title can start it at 480i. In my case, this happened with a Super Mario 64 romhack called Usamune.

This problem is mitigated by setting only the video mode to NTSC via Anyregion Changer, on an otherwise PAL-configured Wii.

**Version**
0.10.0

**Expected behaviour**
The channel should start at 480p as per the system menu setting.

DacoTaco, 2024-10-25:
what is the video mode priiloader was outputting? 480p or 480i?

pyorot, 2024-10-28:
sorry it might take me a few days to confirm this, because i don’t have any batteries on me and need them to menu thru ARCME to set the video mode back to PAL

pyorot, 2024-11-04:
> what is the video mode priiloader was outputting? 480p or 480i?

hello hello. i am pleased to inform that i have found batteries and, regardless of the console video mode being PAL or NTSC, Priiloader outputs 480p. This changes to 480i after starting the SM64 romhack if the video mode is PAL, and remains 480p if it is NTSC.

The console is set to 480p in system settings, and I found the switch to 480i happens both with and without the two system menu hacks “480p fix” and “remove deflicker” (or similarly named) enabled. “Region free everything” is also enabled in all of the above.

DacoTaco, 2024-11-04:
thanks for testing! that means i might need to look at my video changing code, if that is even done on title, for out of region stuff. 
also, keep in mind that system menu hacks have no effect on games or titles loaded by priiloader :)

if priiloader does not change video mode for titles, im not sure if this is something i can fix cause that would mean its the title itself that is changing it :)

pyorot, 2024-11-04:
you’re welcome :)

since i think priiloader likely [does set video mode for titles](https://github.com/DacoTaco/priiloader/blob/cc2fccd27128edf8c5bdd34a5cec367621993f29/src/priiloader/source/titles.cpp#L403), am i understanding you correctly that you think maybe the title itself is reverting the video mode to match the console hardware/area/game-region setting?

i know of gamecube games that stubbornly boot in 480i on any-region wiis since the loader forces 480p but the game then reinitialises the video afterwards. but this is an ntsc title starting, on a pal-configured pal-region wii, in pal video mode

if the video mode can’t be forced correctly then it may be an argument to add the feature to priiloader to change the video mode persistently in the wii’s setting.txt. one possible system is priiloader setting this to a native setting before the wii displays anything on boot, and then setting it to match a disc/title before launching it

i’d like to remove the need to use anyregion changer between game launches just for playing games. also i believe somebody told me he got an sm64 title to start in 480p via a usb loader app. i don’t believe such an app would use game-specific hacks but who knows

eku, 2024-11-04:
> i don’t believe such an app would use game-specific hacks but who knows

I would say, yes, they patch the code.

DacoTaco, 2024-11-14:
> you’re welcome :)
>
> since i think priiloader likely [does set video mode for [...]

@eku is right, loaders generally patch games *a lot* , even titles and disc games. from what i can see the code already reset/shutsdown VI if the title is from another region than what priiloader detected from settings (aka region from wii). the title should then reinit the video afaik. i dont know if i should shutdown the VI/video or force load different mode, thats something somebody from a ~~piracy~~ backup loader would know. i will not be modifying the wii's setting.txt, as that makes it very easy to brick a wii or lock out a user. other apps might do it, i have no control over that, but i will not do it.

for Disc games priiloader already switches video mode before launching the disc btw, because some SDK titles refuse to boot otherwise, but it does *not* change the wii settings.

pyorot, 2024-11-15:
> @pyorot : do you have bootmii @ boot2? i have an installer ready that sets video mode before launching a title, see if that works : [boot.zip](https://github.com/user-attachments/files/17758124/boot.zip)

thanks, i can check it out. i don’t have bootmii @ boot2 on my disposable wii; i’d have to install this and some title wads onto my original wii if i need that. is it needed? ordinary priiloader installed fine for me with bootmii as ios

DacoTaco, 2024-11-15:
> thanks, i can check it out. i don’t have bootmii @ boot2 on my disposable wii; i’d have to install this and some title wads onto my original wii if i need that. is it needed? ordinary priiloader [...]

its not needed, but its just a safety precaution. feel free to install it on the wii, just know i didnt install it on mine to verify but im positive it'll be fine :)

pyorot, 2024-11-15:
no luck, still 480i. i tested what i think is an original sm64 ntsc-u title (no romhack) on the pal wii and it was still 480i on this priiloader version. there was also a regression where the names all showed up as ???????? tho and i had to navigate via the bracketed game IDs

it was also a journey to get this priiloader version installed lol; the installer wanted a patched ios 36 and was extremely picky about which one. it’s my first time hearing the term CIOSPAGHETTI lmaooo

DacoTaco, 2024-11-15:
thats because your meta.xml was missing or incorrect and it didn't get full access from HBC.
sad to hear the video fix didn't work. if that doesnt work, there is not much i can do sadly.

pyorot, 2024-11-16:
Could you show me the patch that you sent me to test please? I’m wondering about the meaning of some parts of the code, like say the explicit setting of NTSC interlaced mode [here](https://github.com/DacoTaco/priiloader/blob/cc2fccd27128edf8c5bdd34a5cec367621993f29/src/priiloader/source/titles.cpp#L428) that gets passed to VIDEO_Configure() later in the code.

pyorot, 2024-11-16:
@DacoTaco OK so, after reviewing some code, I suspect that the problem is this line of code I linked to in my previous reply, which forces interlaced for NTSC. At the very least, it's the primary obstacle before any other problems.

Here's some code excerpted from USBLoader GX from its channel-loading path, which I checked is able to set 480p in SM64 and doesn't appear to use any game-specific patches to do so:

```c
#define	 Video_Mode		((vu32*) 0x800000CC)

{
	rmode = VIDEO_GetPreferredMode(0);

	/* Get video mode configuration */
	bool progressive = (CONF_GetProgressiveScan() > 0) && VIDEO_HaveComponentCable();

	switch (videoselected)
	{
		case VIDEO_MODE_DISCDEFAULT: // DEFAULT (DISC/GAME)
			/* Select video mode */
			switch (diskid[3])
			{
				// NTSC
				case 'E':
				case 'J':
					rmode_reg = VI_NTSC;
					rmode = progressive ? &TVNtsc480Prog : &TVNtsc480IntDf;
					break;
			}
			break;
		break;
	}

	/* Set video mode register */
	*Video_Mode = rmode_reg;
	DCFlushRange((void *) Video_Mode, 4);

	/* Set video mode */
	if (rmode != NULL)
		VIDEO_Configure(rmode);

	/* Setup video  */
	VIDEO_SetBlack(TRUE);
	VIDEO_Flush();
	VIDEO_WaitVSync();
	if (rmode->viTVMode & VI_NON_INTERLACE)
		VIDEO_WaitVSync();
	else while(VIDEO_GetNextField())
		VIDEO_WaitVSync();
}
```

I think the user experience we're looking for is to pull the Wii system settings' 480p mode out of ` VIDEO_GetPreferredMode(0)` and use that to choose between `&TVNtsc480Prog` and `&TVNtsc480IntDf`. At least, I don't think defaulting to, let alone forcing, interlaced makes sense anymore in 2024.

I'm curious what other change you made and gave me to test, and look forward to hearing your ideas.

pyorot, 2024-11-18:
> @eku is right, loaders generally patch games _a lot_ , even titles and disc games. from what i can see the code already reset/shutsdown VI if the title is from another region than what priiloader [...]

just a quick note to clarify that shutting down video (previous behaviour) actually worked, but just wasn't being called cos of an unrelated bug. i haven't yet checked if shutting down then force loading a different mode also works, but will edit that into this post later. and which should be done (assuming it works), idk

PR #379 (2024-11-21): viTVMode access macros + bugfixes
A fix related to [libogc video_types.h](https://libogc.devkitpro.org/video__types_8h.html). Thruout the code, the whole of `GXRModeObj.viTVMode` was being erroneously compared to `VI_NTSC`, which is a VI_TVMODE_FMT rather than a full VI_TVMode. This patch adds two macros to invert the `VI_TVMODE(fmt, mode)` transformation and recover the FMT and the MODE (this should be upstreamed to libogc), and then applies the macros thruout the codebase. The changes are pretty much:
1. `rmode->viTVMode == VI_NTSC` → `VI_TVMODE_FMT(rmode->viTVMode) == VI_NTSC` (bugfix)
2. ` [... cut]
closed 2024-11-21 by DacoTaco