DacoTaco/priiloader #380: "Dump Gecko output" setting not working

pyorot, 2024-11-20:
**Describe the bug**
Having this setting on and rebooting the Wii a few times, it never creates the file `prii.log` on the root of the removable device, whether SD card or USB drive, when only one is plugged in. I tested it by launching titles with the menu showing ???????? for every title name, which should log something.

*continued from [this discussion](https://github.com/DacoTaco/priiloader/issues/379#issuecomment-2487714188_)*

decided to file issue cos tbf, i think i know what causes this. i decided to rewrite [wii swiss booter](https://github.com/pyorot/wii-swiss-booter/) yesterday on a whim to get errors printed. i found that devices will never start up unless [`fatInit` is run first](https://github.com/pyorot/wii-swiss-booter/blob/main/src/device.c), which auto-creates mount-points for wii sd and wii usb that i have to then get rid of cos i don't need them. this is clearly a change in the libraries cos the original wii swiss booter code from 2011 compiled fine but then systematically failed to load swiss while being installed side-by-side with a [version built in 2022](https://github.com/RibShark/Wii-Swiss-Booter-480pFix). i could check by adding just the `fatInit` call to the 2011 version and testing that. idk which libraries the 2022 one was built with but it at least makes it likely there's been a regression in the past 2 years

but yeah, someone else testing a current build of priiloader should clarify this quickly

DacoTaco, 2024-11-21:
thats odd, because for me it still just works with my sd card and usb stick. no need for fatInit

DacoTaco, 2024-11-21:
found the issue, priiloader didn't init libfat at all which shouldn't have worked, but used to work haha
and now with libdvm, it was really saying fuck you haha.
my mistake, its now working again

closed 2024-11-21 by DacoTaco

pyorot, 2024-11-21:
> found the issue, priiloader didn't init libfat at all which shouldn't have worked, but used to work haha and now with libdvm, it was really saying fuck you haha. my mistake, its now working again

nice one, ye. that's what i meant, the `fatInit` wasn't needed, now it is. but is there a way to do the init without auto-mounting anything? i need devices to mount to the same handle one-at-a-time in that other project