DacoTaco/priiloader #381: can you confirm if priiloader will still display if wifi module is bad

stanwebber, 2025-01-26:
i've tried to research this, but i can get no definitive answer. i recently replaced the wifi module on my wii, but i still get a black screen on boot. i have reason to believe the replacement wifi module i received was used, but before i order another i would like to have some confirmation the wifi module is still the issue. i can rule out the bluetooth module as that was successfully replaced prior.

priiloader will not display when holding the reset button. can someone please confirm if a bad wifi module will prevent priiloader from displaying (the same way the system menu is affected). this would at least give me some confidence to proceed with replacing the wifi module again.

AndrewPiroli, 2025-01-26:
I've tested this, Priiloader will not work without a WiFi module. Almost nothing works without a WiFi module.

stanwebber, 2025-01-26:
thanks for the confirmation. this is what i was hoping to hear.

in my research i came across the rvloader project which utilizes the priiloader load/install feature. rvloader has a no-wifi option for modders that want to completely eliminate the wifi module so priiloader must be functioning on some level even if it can't display. i was hoping to use rvloader as a diagnostic tool, but i can't figure out how to retroactively insert it if the wifi module is already bad/absent.

AndrewPiroli, 2025-01-26:
To use RVLoader with no wifi requires a patched IOS to be installed ahead of time that prevents it from trying to talk to the WiFi module. It's not something that you can do after the fact without getting into NAND modifications.

The issue is IOS tries to set up networking and it was programmed to just assume the WiFi module is present and working at all times. If it's not IOS crashes/hangs forever. Pretty much all homebrew other than BootMii is built on top of IOS, so everything just stops.

DacoTaco, 2025-01-26:
what andrew said is true. on top of it, priiloader is part of system menu when it is installed, and System menu is booted by IOS80. so if IOS80 just hangs on startup and not boot system menu, priiloader is never launched.

though ive never confirmed how far it gets in the boot process myself, i assume that is what happens but no idea why or where in ios code that happens :)

Masamune3210, 2025-01-26:
likely early on in hardware initialization stuff. If I remember correctly, the issue is that it tries to set the adapter up and part of that is initializing it and then waiting for a response from it before continuing. No/Faulty Adapter = Infinite Wait for a message that will never come.

DacoTaco, 2025-01-27:
> likely early on in hardware initialization stuff. If I remember correctly, the issue is that it tries to set the adapter up and part of that is initializing it and then waiting for a response from it before continuing. No/Faulty Adapter = Infinite Wait for a message that will never come.

thats what is expected ye. but i never checked the code why the cpu wouldn't just move to a different module and keep processing there, or if the module crashes and takes down the whole IOS (it can happen). 

anyway, i think that is a solid: maybe the 'new' wifi module is also dead or something else killed the wii ^^

closed 2025-01-27 by DacoTaco