DacoTaco/priiloader #382: Shutdown to option turns off Wii despite setting it to Priiloader (or Autoboot)

aop-runs, 2025-02-01:
**Describe the bug**
After changing the "Shutdown to" option in Priiloader, my Wii will always power off completely instead of taking you back to Priiloader or any application stored in Autoboot. Same thing happens regardless where you power off. (System Menu, Title, HB App, etc.)

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Priiloader by any means
2. Select Settings
3. Change "Shutdown to" option to Priiloader (or Autoboot)
4. Save Settings
5. Access anything of your choice on the Wii
6. Press the power button on the Wii

**Version**
0.10.0

**Expected behavior**
When pressing the power button, it should go back to Priiloader rather than turn off

**Additional context**
My Wii only has boot1 installed plus Autobooting wherever doesn't change anything. Wii is patched with cIOS files from official HB guide, but the issue occurred with vanilla IOS for me as well.

Masamune3210, 2025-02-01:
Not going anywhere fast without boot2 and a whole lot more lol

Joking aside, I'm assuming you mean you have bootmii installed as boot2?

aop-runs, 2025-02-01:
I didn't have the option for boot2 so bootmii as IOS is what I have (if that's the correct name)

Masamune3210, 2025-02-01:
Ok, just making sure. The difference between the two can matter quite a lot

DacoTaco, 2025-02-08:
i just tested this by setting the option, going to SM and pressing power. 
this booted priiloader, so it works for me.

what is running when you press the power off button? have you tried booting System Menu and pressing power?

FipekPL, 2025-02-08:
I actually face the identical issue on my wii. It doesn't matter from which screen I press the Power button the WII always shutsdown. System Menu/Homebrew/Inside a game.

aop-runs, 2025-02-08:
HB Channel + Apps and Virtual Console titles is mainly what I use so that's where it happens. Booting to System Menu will also shutoff completely when pressing power.

DacoTaco, 2025-02-09:
freaking weird that it works fine on my wii, the one wii it *should* break on haha

if you don't mind, i would like to share a few priiloader installers that will install priiloader versions that have things changed, so i can hopefully debug the issue.

we will start with the first one: the latest version that is up to date with all code fixes and changes : 
[boot.dol.zip](https://github.com/user-attachments/files/18723867/boot.dol.zip)

place, and overwrite, the boot.dol on your SD (apps/priiloader) with this boot.dol.
remove all usb devices from the wii and install priiloader using this installer.
then try again

FipekPL, 2025-02-09:
Thank you for the new file and help!
I just had some time to test it with the new boot.dol and it's sadly the same result. 
The Autoboot: and Return to: settings work as expected with any selection value. But the Shutdown to: always shuts down my Wii regardless of what I selected. (I tested from the HB channel, System Menu and a WiiVC game)

DacoTaco, 2025-02-09:
@FipekPL : do you have boot2 @ bootmii by any chance?

FipekPL, 2025-02-09:
No I have only boot1 in bootmii. (its a newer Wii according to Syscheck Boot2 v4)

DacoTaco, 2025-02-09:
that sucks. im not 100% comfortable with the next build, it changes things a bit and if it crashes...

If you want to risk it, this mounts the SD a bit earlier, before it decides what to do with the bootstate, so it can be logged.
Could you remove all USB devices, enable the dump debug setting and shutdown the wii when inside System menu? on the sd there should be a Prii.log that i would love to see its input of.

but please be careful with this build if you install it, and install regular priiloader back after the test.

[boot.dol.zip](https://github.com/user-attachments/files/18724987/boot.dol.zip)

FipekPL, 2025-02-09:
I am out for the rest of the day but I can test it tomorrow and send you the Prii.log.

FipekPL, 2025-02-10:
@DacoTaco 

OK so I just had some time to test with the last build you attached. (Booting to System menu then pressing the Power button on the Wii with logs enabled).
The behavior is still the same, basically nothing has changed (don't know if I make something wrong during installation).
I tested it twice and double checked my settings (Shutdown to: Priiloader).

I also included my System Menu Hacks just in case.

[prii.log](https://github.com/user-attachments/files/18738232/prii.log)

[image]

[image]

[image]

[image]

DacoTaco, 2025-02-10:
@FipekPL : it did exactly what i was hoping to see. I needed that Bootstate log line in your log to figure out what its doing.
However, it has set bootstate 3 which is restart/reboot not shutdown ( 5 ) ...

any information on the wii you have? is this a regular wii? wii mini? vwii ?
is WC24 setup, any changes to settings/shutdown behaviour? 

also, this is not related to your shutdown issue @Ingunar i think, since you got 255 which is invalid normally

FipekPL, 2025-02-10:
@DacoTaco 

Its a "normal" wii (red case RVL-001(EUR)) which I just bought off ebay.
I installed some cIOS for USB Loader GX maybe thats an issue?

Here the full syscheck info for my system

[SysCheck.csv](https://github.com/user-attachments/files/18739303/SysCheck.csv)

WC24 isn't setup I get the following info in settings

 
[image]

aop-runs, 2025-02-10:
Sorry for not contributing as much as I was busy yesterday. I also have a normal red wii (mine is the 25th anniversary one) and my SysCheck is similar to FipekPL's wii. Could be a trend of which wiis may have this issue, but can't really vouch beyond that. Don't have WC24 set up, but do have a Internet connection to use some HB apps.

[IOSsyscheck.log](https://github.com/user-attachments/files/18741934/IOSsyscheck.log)
[SysCheck.csv](https://github.com/user-attachments/files/18741933/SysCheck.csv)