DacoTaco/priiloader #385: Include a WUHB forwarder that's TV only?

PlayerGameSK, 2025-02-24:
Yeah so basically I've been using the included WUHB forwarder to get into Priiloader on my Wii U with Aroma, however I noticed that it defaults to having both the gamepad and TV screens on, even though I never use the gamepad while playing Wii games, thus it just gets its battery drained for no reason while being stored away.

Would it be possible to include a WUHB that defaults to TV only? Thanks.

Ingunar, 2025-02-24:
There is no need to have DRC-only and TV-only builds since they are automatic and based on connected devices.

In your case, you can override all WUHB forwarder settings to be TV-only with [WiiVCLaunch](https://github.com/Lynx64/WiiVCLaunch)

PlayerGameSK, 2025-02-24:
Yeah, no, that does not work, I have WiiVCLaunch installed and set to TV Only, yet the forwarder still launches on both TV and Gamepad.

closed 2025-02-24 by DacoTaco