DacoTaco/priiloader #389: Error building latest code.

impeeza, 2025-04-13:
**Describe the bug**
Building latest code (2025_04_13) ends with the error

```
OpenDolBoot ...
In file included from source/NandLoaderInjector.cpp:30:
source/../include/dolHeader.h:28:10: fatal error: arpa/inet.h: No such file or directory
   28 | #include <arpa/inet.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
In file included from source/OpenDolBoot.cpp:31:
source/../include/dolHeader.h:28:10: fatal error: arpa/inet.h: No such file or directory
   28 | #include <arpa/inet.h>
      |          ^~~~~~~~~~~~~
compilation terminated.

```

**To Reproduce**

I use MSYS's MinGW64 with **all** DevKitPro packages installed

For building I used the commands:

```
pacman -Syuu --needed --noconfirm git make wii-dev mingw-w64-x86_64-gcc
cd ~
rm -rf ~/priiloader
git clone --recursive https://github.com/DacoTaco/priiloader
cd ~/priiloader
make all

```

**Screenshots**
A searching of `arpa` on my MSYS folder
[image]

Error on building
[image]

I am Missing any library?

Aeplexi, 2025-04-13:
devkitpro works better on WSL, try using it?

impeeza, 2025-04-14:
WSL is not good for me.

I switched to MSYS (instead of MinGW) and after installing `msys2-runtime-devel` and now the error is:

[image]

I don't know where the `uinit` should be defined.

will try a Ubunto Virtual Machine later.

impeeza, 2025-04-14:
sorry closed the issue by error.

urherenow, 2025-04-14:
Not at home right now to test myself, but thought I should mention 2 things.

1) ALWAYS use msys2, unless you are specifically compiling a Windows binary. MinGW is required only for that.

2) The configuration is separate, between launching mingw and msys. Reinstall all the things under msys.

impeeza, 2025-04-14:
Thanks a lot,  the code built fine on a Ubuntu VM.

will try your advice later.

Thanks a lot for the quick answer.

impeeza, 2025-04-14:
I really don't know why I continue pressing the close issue instead of just comment!!!!

I found:

if I install all requeriments on MSYS environment, and run the `make` and also on the file `priiloader/tools/OpenDolBoot/source/OpenDolBoot.cpp` I add the line `typedef unsigned int uint;` at start (I did on the line 36 after all includes,  the build process works more than fine.

[image]

I am not a great coder but the idea of add that comes from some forums about the changes between C11 and c17 and how some compilers have the `uint` implicit defined and some others need the user to define it on the code.

what do you think about the inclusion of that line?

by the way I build the code with that line included on the Ubuntu VM and also build fine.

DacoTaco, 2025-04-14:
i would need to go into windows and build under msys, but Opendolboot is a tool that should build using gcc of the host, not devkitpro's toolkits. if its looking for uint or the arpa header, it should find the host's gcc headers and not devkitppc's.

afaik msys should ship with gcc of the chosen environment though...

impeeza, 2025-04-14:
> afaik msys should ship with gcc of the chosen environment though...

Yes, you are right, on MSYS is embedded on the MSYS environment, I am still looking if there is a package on MinGW.   the problem about the ARPA/INET.h  was resolved using MSYS.

but now the problem about the `uint` arrised, seems on linux gcc the definition of `uint` is already defined but on tcc and MSYS is not and you must define it on the code.

DacoTaco, 2025-04-14:
ha, ye. seems uint is not standard. my bad
https://stackoverflow.com/questions/3552094/c-uint-unsigned-int-int

ill change it to unsigned int tonight

DacoTaco, 2025-04-14:
@impeeza : could you check if `uint32_t` works/exists in msys?

impeeza, 2025-04-14:
> `uint32_t`

Seems yes, changing the `uint` for `uint32_t`

build without problem

[image]

eku, 2025-04-14:
> ha, ye. seems uint is not standard

Use <stdint.h> for types.

impeeza, 2025-04-14:
by the way, this is the explanation of MSYS environments

[image]

https://www.msys2.org/docs/environments/

impeeza, 2025-04-14:
> > ha, ye. seems uint is not standard
> 
> Use <stdint.h> for types.

Nop stdint.h on MSYS do not contains `uint` but it does contains `uint32_t`

eku, 2025-04-14:
> > > ha, ye. seems uint is not standard
> > 
> > 
> > Use <stdint.h> for types.
> 
> Nop stdint.h on MSYS do not contains `uint` but it does contains `uint32_t`

I meant use only types from stdint.h.

DacoTaco, 2025-04-14:
fixed on main ^^;
sorry for that

closed 2025-04-14 by DacoTaco

impeeza, 2025-04-14:
There is no need to be sorry, you are doing a great work.  regards mate.

I will open another issue about building LoadPriiloader