DacoTaco/priiloader #392: Cannot autoboot HBC

60fpshacksrock, 2025-05-21:
**Describe the bug**
PL will not autoboot HBC anymore

**To Reproduce**
Install priiloader and set it to boot The Homebrew Channel

**Version**
Tried 0.8 and 0.10

**Expected behavior**
Homebrew channel should be autobooted

**Screenshots**
[If applicable, add screenshots to help explain your problem.]

**Additional context**
After accidentally deleting HBC while meaning to delete the save data to remove the custom theme and trying and install both homebrew channel version 1.1.0 and 1.1.2 it will not autoboot them. It will autoboot hbc 1.1.4, but that version crashes upon reading my SD card and likely trying to display apps and their icons.

DacoTaco, 2025-05-22:
normally, multiple HBC installs on a single wii makes no sense. System menu even removes the older installations it knows...
@60fpshacksrock : why do you have multiple versions installed and i think you'd rather figure out why 1.1.4 crashes on your sd and what file breaks it. hell, maybe even format the SD and place all the files back on...

Naim2000, 2025-05-22:
Confused.
* Is it not autobooting HBC, or is HBC crashing upon trying to read your SD card? Take out the SD and test.
* Is this only happening to 1.1.4? What about the 1.1.0 and 1.1.2 that you.. said you have? (though, past this point, this isn't relevant to "Priiloader" anyways)

60fpshacksrock, 2025-05-22:
@DacoTaco
I did have multiple hbc versions installed, by accident. But if I was not clear enough, I removed them all, re-installed system menu 4.3 and afterwards tried different versions without any other versions installed and still get the same error.

I even uninstalled and re-installed priiloader 0.10. It will not autoboot any version with even only one version installed.

DacoTaco, 2025-05-22:
@60fpshacksrock : this is a 2 part issue. the HBC detection code in priiloader has a bug that was fixed recently which is why its trying to boot 1.1.4 .
however, it is recommended to use 1.1.4 regardless.

we also suspect there is an app on your SD that has invalid data in it's meta.xml. because of that it's recommended to get all the data of your SD, reformat it, and put the data back, one app at a time and testing it after every app to find which meta.xml has invalid data.

60fpshacksrock, 2025-05-22:
I will see what I can do.

DacoTaco, 2025-05-22:
let me know if you find anything!
there are some ideas as to what might be wrong if you do find the app :)

60fpshacksrock, 2025-05-23:
I am still checking, (was gone earlier) but I am now back and so far I have found at lease one that causes a crash; DOSBox Wii

Here is the xml (in quotes), it may be the spaced apart text.

"<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<app version="1">
  <name>DOSBox Wii</name>
  <coder>Tantric</coder>
  <version>1.7</version>
  <release_date>20120630</release_date>
  <short_description>DOS Emulator</short_description> 
  <long_description>Port of DOSBox to the Wii</long_description> 

<ahb_access/>
</app>
 "

DacoTaco, 2025-05-23:
its better to upload the full file instead of copy pasting as invisible characters (like null characters) might be the reason it broke hehe

also, official HBC might be considered dead, but replacement ideas are floating around.
i think in general the wii just needs a good GUI lib or somebody that knows what they are doing in SDL haha

60fpshacksrock, 2025-05-23:
It all okay now. It was apparently malformed xml's. Sometime either an empty space at the end on a space behind the long_description part.

DacoTaco, 2025-05-23:
good to hear its fixed! sorry that the priiloader & HBC bug has caused so much headaches!

closed 2025-05-23 by DacoTaco