DacoTaco/priiloader #393: Priiloader cannot find hacks_hash.ini even though it's on the USB stick

ImDaNintendo-WiiGuy, 2025-05-23:
So, I bought a Wii, put Priiloader on it (only to find out it was already installed), tried to open the "System Menu Hacks" section and it said "Can't find device apps/priiloader/hacks_hash.ini" and "Can't find hacks_hash.ini on NAND" I even tried downloading the file from this GitHub repo, it just didn't work. (because i put the file in apps/LoadPriiloader/)

**To Reproduce**
Steps to reproduce the behavior:
I don't really know how to reproduce this glitch. I guess i have to put the file in D:/apps/priiloader and not D:/apps/priiloader/apps/priiloader/

**Version**
IOS58, 4.3U. I lately installed IOS56 because of the WiiMart Wii Shop Channel revival project. YAWM ModMii WAD Manager still said i had IOS58, but WiiMart still worked.

**Expected behavior**
I expected it to just load the system menu hacks menu straight up lol-

**Screenshots**
[image]
[image]

**Additional context**
How do i fix this?

DacoTaco, 2025-05-23:
is an sd card inserted?
priiloader will take the SD card as priority one, than check usb, than nand.
if it couldn't find the file it means it either wasn't in (or couldn't access) sd:/apps/priiloader directory, or it wasn't in (or couldn't access) usb:/apps/priiloader to look for the hacks_hash.ini file.

please verify the file is in the correct place. i'd also recommend putting it on the SD if there is one, because USB sticks and drives are unstable

ImDaNintendo-WiiGuy, 2025-05-24:
@DacoTaco no, there is no sd card inserted, i don't have a supported wii sd card, only a microSD one, i only have an usb.

DacoTaco, 2025-05-25:
does load/install app work and list all applications installed on the usb?