DacoTaco/priiloader #395: Bypassing the system menu on boot causes font issues on Japanese systems

AuroraWright, 2025-06-08:
**Describe the bug**
If Priiloader is set to autoboot a dol or the Homebrew Channel, and then Wii/Gamecube games are launched at any point from there (eg Gecko OS or any other loader), some games exhibit strange behavior that seems font/language related. As an example:
- (Gamecube) Paper Mario RPG (The thousand year door) in Japanese shows a red Nintendo logo instead of blue and then crashes abruptly with a textbox saying "no message" in English
- (Wii) Kororinpa 2 shows question mark instead of text messages throughout the game
If the System Menu is loaded at any point the issue is then fixed until console restart.

**To Reproduce**
Set Priiloader to boot from HBC and load an affected game in some way 

**Version**
0.10

**Expected behavior**
Games should work without issues

AuroraWright, 2025-06-08:
All I could find that is probably related: https://gbatemp.net/threads/gibberish-text-in-japanese-games-on-american-wii.381402/#post-5339312

DacoTaco, 2025-06-08:
@AuroraWright : i assume it also happens if priiloader launches the game from the menu?

DacoTaco, 2025-06-08:
possible fix: set register on priiloader boot & when booting a game : https://wiibrew.org/wiki/Hollywood/Registers#HW_VI1CFG

DacoTaco, 2025-06-08:
maybe related to #343

DacoTaco, 2025-07-13:
fixed in #402

closed 2025-07-13 by DacoTaco