DacoTaco/priiloader #396: vWii NTSC-K Discs Freeze Wii Menu

ToadetteHackFan, 2025-06-09:
**Describe the bug**
Attempting to read NTSC-K discs (at least it does with my korean Mario Kart Wii), it freezes the entire Wii Menu

**To Reproduce**
Enable Region Free EVERYTHING and insert a korean disc. Wait a few seconds and it froze.

**Version**
v0.10

**Expected behavior**
Be able to load the game just like it can on my regular Wii or say that the game disc cannot be read

**Additional context**
I am aware that the Wii U never had any NTSC-K support so getting this to work may be hard or impossible. I cannot test any other disc as this is my only korean disc. The disc seems to work fine considering my Wii can read it and I can make a dump with my Wii U.

ToadetteHackFan, 2025-06-09:
I should note that US and JP copies work perfectly fine on my PAL vWii

DacoTaco, 2025-06-09:
> If I had to guess, the system menu not having the Korean font?

thats why i shared it on discord :p

> I dont think the font matters as I managed to get it to load on my Wii that is also PAL

same game? that is kinda odd if it does run on 4.3E on a pal wii but not pal vWii on pal WiiU

ToadetteHackFan, 2025-06-15:
So, I think I figured out what is causing this. Before buying the Korean Disc, I installed IOS46 to my regular Wii. This of course allows the game to get detected. I did not install IOS46 to my vWii (and definitely wont) and my other Wii. Both have the freezing issue. I had Region Free EVERYTHING and force Games to run under cIOS249 enabled, however I did not force the System Menu itself to run under cIOS249. Doing so on my other Wii fixed the softlock. Once I get to it, I will see if that exact thing is also causing my Wi U to crash. I will close this when it works then. Absolutely my fault in this case here

DacoTaco, 2025-06-15:
ah thats good to know. keep us updated on the results of the tests!

ToadetteHackFan, 2025-06-16:
I just left my home :/. Will do tomorrow

ToadetteHackFan, 2025-06-16:
When I am home, I will also make a SysCheck for you to look at

ToadetteHackFan, 2025-06-17:
[image]

After trying to force it into cIOS249, I got this crashdump and my console never booted back into the vWii Menu anymore. I managed to get into the Homebrew Channel but I still cannot access my System Menu even through exiting the Homebrew Channel

ToadetteHackFan, 2025-06-17:
I did not do any changes yet. I also cannot access homebrew anymore. This version is very unstable

ToadetteHackFan, 2025-06-18:
[SysCheck.csv](https://github.com/user-attachments/files/20801029/SysCheck.csv)
[IOSsyscheck.log](https://github.com/user-attachments/files/20801028/IOSsyscheck.log)

Here is the SysCheck dump

ToadetteHackFan, 2025-06-19:
I have read deeper into the issues tab and noticed that you can load a different version of priiloader than what is installed? Were I supposed to do that? If that is the case, how do I do that?

ToadetteHackFan, 2025-09-14:
I decided to test this on my other Wii now as well that barely had any modifications done to it besides Homebrewing and installing CTGP and Riivolution. v0.10 definitely has issues with running your SM under a different IOS because it also didnt work on that console. I can also confirm that my Wii where it worked initially was able to do the SM IOS patch due to an older version of Priiloader (0.9.X) and that stopped working with v0.10 on that console as well. So this is definitely not a vWii/Wii U specific issue