DacoTaco/priiloader #399: Priiloader Wiimmfi Error 20100

GeorgeGamez, 2025-07-03:
**Describe the bug**
Wiimmfi setting in Priiloader gives error 20100

**To Reproduce**
Steps to reproduce the behavior:
Enable Wiimmfi Auto-Patch in Priiloader Hacks Menu
Open MarioKart Wii
Attempt to Connect to the Internet.

**Version**
0.10.0

**Expected behavior**
Connects to the Internet and Continues to Next Menu

**Screenshots**

[image]

[image]

**Additional context**
I have tested this with another Wiimmfi patcher, and I don't get an error with them.

Leseratte10, 2025-07-04:
Unlikely. I can see two people online with the Priiloader patcher right now, and successful connections to Wiimmfi using this patcher from 10+ different people today. There's definitely not a general issue with the patcher that applies to everyone.

DacoTaco, 2025-07-04:
> Unlikely. I can see two people online with the Priiloader patcher right now, and successful connections to Wiimmfi using this patcher from 10+ different people today. There's definitely not a general issue with the patcher that applies to everyone.

any different reason that could cause this then? @GeorgeGamez mentions regular patcher works fine...
@GeorgeGamez : you did save the hacks after enabling it right? do other games work?

GeorgeGamez, 2025-07-04:
When I tested this, I had previously tested the internet connection. I also launched it through the disk channel. I will test the online update blocker. I have not yet tested it with a different game.

GeorgeGamez, 2025-07-14:
I tried going to the update menu. It doesn't seem blocked and it continued to this menu. I don't know when the block is supposed to happen, but I could reach this menu.

[image]

GeorgeGamez, 2025-07-14:
I just tried a different game. It successfully connected when I used Tetris Party Deluxe.