DacoTaco/priiloader #403: Region Free EVERYTHING not working on korean consoles

ToadetteHackFan, 2025-07-19:
**Describe the bug**
I recently bought a korean wii to be able to test my modpack on NTSC-K on console. It was region changed from K to U very poorly. I reinstalled every IOS (using modmii) and changed it back to korea. Some channels are still not korean. Setting Region Free EVERYTHING to true does not do anything on the System Menu (other region games will not show up). Using Riivolution to bypass the region locks does work, however causes crashes when Region Free EVERYTHING is enabled at random points.

**To Reproduce**
Steps to reproduce the behavior:
1 - Somehow get a korean console.
2 - Enable Region Free EVERYTHING
3 - Boot the system Menu
4 - Throw in a non korean game and see if it loads up

**Version**
v0.10 Kappa

**Expected behavior**
Show the other region games and not cause weird crashes in games from other regions

**Additional context**
Like I said, this console used to be region changed to US. For whatever reason it didnt have english as a language. Instead, it had Chinese. I region changed it back to korea. System Menu is in korea now, some channels however are not. Playing Mario Kart Wii (NTSC-J) on this korean console causes the game to crash after the countdown when Region Free EVERYTHING is enabled.

DacoTaco, 2025-07-19:
isn't this a duplicate of your other issue #396 ?

ToadetteHackFan, 2025-07-19:
No, this time priiloader has issues on a korean console, not playing korean MKW on my Wii U