DacoTaco/priiloader #404: Add proper support for dev-keyed consoles

GerbilSoft, 2025-07-26:
**Describe the bug**
Priiloader can be installed on dev-keyed consoles (CAT-I, CAT-R, CAT-DEV) once Wii mode is installed properly, but it cannot launch the System Menu because it has a hard-coded retail ancast key.

**To Reproduce**
Steps to reproduce the behavior:
1. Install priiloader on dev-keyed console.
2. In priiloader's menu, attempt to start the system menu.

**Version**
Priiloader 0.10.0

**Expected behavior**
Wii System Menu starts.

**Screenshots**
[image]

**Additional context**
Most dev consoles don't have Wii Mode, and up until now it wasn't possible to get it working. I'll be writing up a post on my personal site detailing how to set it up.

GerbilSoft, 2025-07-26:
I believe I have the code modified correctly, but the .dol I built locally isn't loading in Homebrew Channel for some reason. (It stops at 6%.)

The .elf version did load with Wiiload, but when I try to load System Menu from Priiloader, it crashes with a DSI exception, so I'm pretty sure something went wrong.

Commit: https://github.com/GerbilSoft/priiloader/commit/d9f4f405a78dca01ea5af675c7627ace92c602b7

DacoTaco, 2025-07-27:
DSI is indeed means something went wrong :p

anyway, good change & it doesn't look wrong to me tbh.
got an idea where it crashed, from using read-elf/addr2line and the elf?

GerbilSoft, 2025-07-27:
Crash screen:
[image]

Using addr2line on the stack dump, I get:

```
0x81354070
/home/davem/projects/devkitpro/pacman-packages/libogc/src/libogc-2.12.1/wiiuse/wpad.c:314
0x813545e8
/home/davem/projects/devkitpro/pacman-packages/libogc/src/libogc-2.12.1/wiiuse/wpad.c:836 (discriminator 1)
0x81348a44
priiloader/src/priiloader/source/SystemMenu.cpp:383
0x81345df4
priiloader/src/priiloader/source/main.cpp:2704
0x81331824
??:?
0x8136c4e4
/home/davem/projects/devkitpro/pacman-packages/libogc/src/libogc-2.12.1/libogc/lwp_threads.c:699
0x8136c48c
/home/davem/projects/devkitpro/pacman-packages/libogc/src/libogc-2.12.1/libogc/lwp_threads.c:220
```

SystemMenu.cpp:383 is ShutdownMounts();, and the line before that is Input_Shutdown(). I suspect this *might* be due to some dkP updates on my system.

DacoTaco, 2025-07-27:
ye, that crash is not related to your change and is something i have still to look into.
what you can try to avoid it is somehow have no wiimotes synced to the wiiu and get it to boot SM.
maybe set it up to boot SM by default, boot vwii and pull out the wiimote batteries after clicking the launch button in wiiu mode.

its on my todo list...

GerbilSoft, 2025-07-27:
I disconnected the Wii Remote right after starting Wii Mode but before it rebooted. System Menu loaded without the Priiloader menu, and it seems to be working: Disc Channel is showing the banner for a EUR RVT-R disc. (Both Wii U Menu and Wii Menu are the USA version.)

Looks like it's working, aside from the input crash.

DacoTaco, 2025-07-27:
then feel free to make a PR! love to see this change :D

GerbilSoft, 2025-07-27:
[image]
Disc is an RVT-R copy of Sonic Colours PAL. (Alas, the contents match the final version, but it's useful for this sort of testing.)

Ingunar, 2025-07-27:
> It seems the Power button can be used for navigating in the Priiloader menu, but Reset is used for selecting an option, and there's no Reset button on Wii Us. Could the Eject button be added as a select function if vWii is detected?

As far as I know, Eject is not connected at all in the Wii mode, we talked about it in Gary's server.

closed 2025-07-27 by DacoTaco